e-m-b-a/emba #392: P61 unblob delivers correct filesystem, but is not used

jblu42, 2022-11-14:
I am tried to analyse an Android QFIL file, which contains one partition of the Android system. The binwalk extractor does not deliver a usable output or extraction of the filesystem, but I was happy to see that the p61_ublob eval actually could extract the filesystem correctly. Problem is, that in the continued analysis steps, the results from the binwalk extractor are used and not the filesystem in the p16 directory.

Any ideas how to fix this?

(Probably Android is out of scope, but due to its similarities with Linux it does deliver quite a good output)

m-1-k-3, 2022-11-14:
The unblob module is currently not completely integrated and the results are not further used. It is an evaluation module to see where unblob performs better. Could you provide the firmware for further evaluation? If you need to analyse the results from unblob you can point EMBA in a second run to the p61 directory as temp solution.

jlucius, 2022-11-18:
While working on the rootfs detection, would that not be a fitting idea? Run rootfs detection on the binwalk directory, if no rootfs found, try rootfs detection on unblob directory, if rootfs found use this one, if not found, still use the binwalk directory with RTOS mode.

m-1-k-3, 2022-11-18:
In the future we will integrate Unblob better into the deep extraction mode. Then it will automatically detect root filesystems in the unblob extraction.

m-1-k-3, 2022-12-04:
last tests are running ...
[image]

closed 2022-12-05 by m-1-k-3