e-m-b-a/emba #400: Docker is missing

torabi12, 2022-11-19:
**Describe the bug**
I wanted to do a clean installation on Kali 2022.3 but the EMBA installer dropped this error at the beginning:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. See error.

**Expected behavior**
Installation starts.

**Desktop (please complete the following information):**
- OS:Kali Linux 2022.03
- EMBA version: Current master branch
- Installation method: default

m-1-k-3, 2022-11-19:
could you do a check on #401?

torabi12, 2022-11-19:
I would try it, but I can clone only the master. Could you help what and how to download?
Sorry I am not good using git.

torabi12, 2022-11-19:
I tried this but I am not sure it was OK:
[image]

Could you help what to download?

m-1-k-3, 2022-11-19:
You need to change into the _installer_ branch or check it out the following way:
```
git clone --branch installer https://github.com/m-1-k-3/emba.git
```

torabi12, 2022-11-20:
Thank you, I tried it and the installation was working fine. Only this minor warning came at the database section:
[image]

I started a scan and I added an existing and empty folder for the log files for EMBA:
[image]

and if the folder exists I always get a question from EMBA to delete the content:
[image]

but the folder is empty. I think if I do not create the folder and I just add a new path to emba starting command this message doesn't come.

torabi12, 2022-11-20:
During the previous scan mongodb went wrong:
[image]

Something is wrong with the Mongo database unfortunately.

[image]

m-1-k-3, 2022-11-20:
> and if the folder exists I always get a question from EMBA to delete the content:
> 
> but the folder is empty. I think if I do not create the folder and I just add a new path to emba starting command this message doesn't come.

As soon as you are using an existing folder for logging EMBA will warn you

m-1-k-3, 2022-11-20:
> Something is wrong with the Mongo database unfortunately.
> 
Could you please check issue #187

closed 2022-11-20 by m-1-k-3