e-m-b-a/emba #405: requests package question

torabi12, 2022-11-20:
**Describe the bug**
During the emba test, the CVE search worked, but after a while the database became inaccessible and CVE-search stopped working.
I tried to follow #187 and I checked the installed requests version:

[image]

but emba uses an older one package:

[image]

Under the scan when CVE-search was working I had this issue:
[image]

maybe it is independent of the CVE problem, but I was not able to finish the scan.

docker-compose gave this message:
[image]

Can I ask for help to get emba to use the latest appropriate request package?

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. Use the firmware available here: https://download.zyxel.com/XGS4600-32F/firmware/XGS4600-32F_4.70(ABBI.4)C0.zip
3. Start EMBA with the following parameters: sudo ./emba.sh -l ~/XGS4600-32F -f ~/470ABBI4C0.bin -p ./scan-profiles/default-scan.emba -X "4.70(ABBI4)C0" -Y "ZxYEL" -Z "XGS4600-32F" -N "470ABBI4C0.bin"

**Desktop (please complete the following information):**
- OS: Kali Linux 2022.03
- EMBA version: actual master branch
- Installation method: default with up to date docker image

m-1-k-3, 2022-11-21:
Looks like there are multiple issues:
* requests package not updated on your host. Could you please try the following: `sudo pip3 install -U requests`
* Some issue with stacs. I will do a test with your firmware.
* mongod not running results in the connection issues from cve-search and the updater. Please do a `sudo service mongod restart` and try again. Also check for open port 27017 via netstat or something similar

m-1-k-3, 2022-11-21:
Should there be a root directory in the firmware? 
[image]

torabi12, 2022-11-21:
Unfortunately I don't know, it's an unknown firmware, I just wanted to test it.
My scan was running for at least 1 hour or more and after that CVE-search went wrong and I stopped it.

I can look at the above tonight when I get home.

torabi12, 2022-11-21:
I did a clean installation and I checked the lines during the emba setup, I found these regarding requests in this order:
[image]

[image]

and after the install I checked it:
[image]

m-1-k-3, 2022-11-21:
> I did a clean installation and I checked the lines during the emba setup, I found these regarding requests in this order

Thanks, thought we fixed this already. I will check this

m-1-k-3, 2022-11-22:
I think you can fix this on your host with the following command:

```
sudo pip3 install --force-reinstall --no-deps --ignore-installed requests==2.28.1
```
I need to build a new docker image to get it also into the container environment.

torabi12, 2022-11-22:
Thank you, the command was working fine!

closed 2022-11-25 by BenediktMKuehne

m-1-k-3, 2022-11-25:
The stacs issue is now also solved here https://github.com/stacscan/stacs/issues/30
I will build a new container the next week and we are good to go.
Thanks for reporting the issues.

torabi12, 2022-11-25:
Great news, thank you!

torabi12, 2022-11-28:
Can I request a message when the container is updated?

m-1-k-3, 2022-11-28:
Should be up

torabi12, 2022-11-28:
After the emba install:
[image]

m-1-k-3, 2022-11-28:
Looks good, doesn't it?

torabi12, 2022-11-28:
Yes, it does:)