e-m-b-a/emba #407: Filter Github Exploit for good output

jblu42, 2022-11-24:
**Is your feature request related to a problem? Please describe.**
Currently when listing exploits in the F20 module a lot of exploits are listed from github that offer no value e.g. collections of vulnerability details that are direct copies from the CVE information, e.g.
https://github.com/thdusdl1219/CVE-Study
https://github.com/facebookincubator/nvdtools
https://github.com/akiraabe/myapp-container-jaxrs
It would be great to see on first view where PoCs are available

**Describe the solution you'd like**
As PoC can appear all over github, probably the best solution would be to blacklist some github pages in this module to allow a better overview

Any other ideas on that?

m-1-k-3, 2022-11-24:
Hi @jblu42 

I'm aware that this feature is currently not ideal. We rely on the trickest collection (https://github.com/trickest/cve) for this feature. If there is some other repository with more accurate details let me know.

jblu42, 2022-11-24:
@m-1-k-3 : It seems they have a blacklist. Does it make sense to propose these pages to their blacklist? I think it will probably get filtered out

m-1-k-3, 2022-11-24:
It would be worth a try.

m-1-k-3, 2022-11-24:
In the meantime we could include a blacklist in EMBA. Could you please create a PR with your blacklisted urls in `config/trickest_blacklist.txt`.
Then I will respect these URLs during building the data file.

jlucius, 2022-11-26:
Will you merge your changes first or shall I just do a PR with the blacklist file?

m-1-k-3, 2022-11-26:
As Benedikt has already approved the PR it is now merged.

m-1-k-3, 2022-11-28:
I did some initial tests with the complete repo name per entry line