e-m-b-a/emba #409: Unable to use full scan profile scans

0xbughunter, 2022-11-25:
While trying to run  scans on a firmware by using full scan profile i get this error

   Qemu system emulator MIPSel - ok
    console.* - not ok
    Missing console.* - check your installation
    busybox.* - not ok
    Missing busybox.* - check your installation
    libnvram.* - not ok
    Missing libnvram.* - check your installation
    vmlinux.mips* - not ok
    Missing vmlinux.mips* - check your installation
    vmlinux.armel - not ok
    Missing vmlinux.armel - check your installation

Note: this happens even after re-installing emba. 
Could anyone from the team please help me out with this?

m-1-k-3, 2022-11-25:
are you running in default mode or full installation?

m-1-k-3, 2022-11-25:
In default mode this error usually means your installation is somehow outdated. Please update your github repo and your docker image:

```
cd emba-directory
git pull
docker pull embeddedanalyzer/emba
```

0xbughunter, 2022-11-25:
i actually did a default installation of emba

m-1-k-3, 2022-11-25:
As the current master branch does not check for ` vmlinux.armel` anymore you are running on an old version of EMBA.
See also here:
https://github.com/e-m-b-a/emba/blob/38a769628fda9ad52c8de90e4249344f146af9a0/helpers/helpers_emba_dependency_check.sh#L527

Please update your installation and it should work again.

closed 2022-12-01 by 0xbughunter