e-m-b-a/emba #419: Alternative PoC data source

m-1-k-3, 2022-12-03:
**Is your feature request related to a problem? Please describe.**

EMBA is currently using the [Trickest](https://github.com/trickest/cve) cve datasource for generating further details on available exploits and PoCs.
We have now seen multiple times that this source is not very clean and a lot of false positives are included. (see also https://github.com/e-m-b-a/emba/issues/407)
@jblu42 has already identified a bunch of false positives and they are already included into the trickest db and additionally in an EMBA blacklist. (see also https://github.com/e-m-b-a/emba/pull/413)

Let's challenge the current approach a bit. Are there any other good and maintained data collections for PoC and exploit code out there.

jblu42, 2022-12-05:
I think the most famous would be exploit-db, but these are already integrated (?).

Still struggling what to do with [https:](https://github.com/ARPSyndicate/cvemon) which currently still is listed in trickest/cve but is a github PoC scanning framework itself and has lots of false positives on trickest/cve. There are also other on github like: https://github.com/nomi-sec/PoC-in-GitHub

There are a few commercial vendors like Rapid7 or Snyk, but that probably does not make sense.

Probably https://packetstormsecurity.com/files/tags/exploit/?rel=nofollow,noopener,noreferrer&target=_blank

Also welcoming good ideas for other sources.

m-1-k-3, 2022-12-10:
> I think the most famous would be exploit-db, but these are already integrated (?).

exploid-db is already integrated.

> There are a few commercial vendors like Rapid7 or Snyk, but that probably does not make sense.

We also included the Metasploit database from R7. 

> Probably https://packetstormsecurity.com/files/tags/exploit/?rel=nofollow,noopener,noreferrer&target=_blank

I will check on this. Thanks for this idea.

m-1-k-3, 2022-12-10:
The Snyk database has also some interesting PoCs included. Is there already an offline version available? See https://security.snyk.io/vuln/SNYK-UNMANAGED-ASTERISK-3160761
Otherwise we should make it available ;)

m-1-k-3, 2022-12-10:
This is a first try to identify PoC code in Snyk advisories: https://gist.github.com/m-1-k-3/267a904687340ddce169782de58a75bb

Probably we need to adjust the PoC detection. Additionally it looks like we need to run it regularly to get the latest data. It looks as they remove older entries from public access.

PR #434 (2022-12-16): Add Packetstorm and Snyk PoC sources
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Feature

* **What is the current behavior?** (You can also link to an open issue here)

Missing Packetstorm and Snyk PoC sources - see https://github.com/e-m-b-a/emba/issues/419

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #419 

include new config files for PoC source:
- config/PS_PoC_results.csv
- config/Snyk_PoC_results.csv

The scripts to create these files are located in the helpers directory:
- snyk_crawler.sh [... cut]
closed 2022-12-16 by m-1-k-3