e-m-b-a/emba #426: Kali Linux 2022.4 tests

m-1-k-3, 2022-12-10:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2022-4-release/
We need to test EMBA on it

**Testcases:**

- [ ] Default/docker installation working (./installer.sh -d)
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
- [ ] Full installation working (./installer.sh -F)
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q
- [ ] Docker build (./installer.sh -D)
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)

torabi12, 2022-12-10:
If I can take part I would do the first scenario, default docker install of EMBA and 2 tests on the latest Kali.

m-1-k-3, 2022-12-10:
> If I can take part I would do the first scenario, default docker install of EMBA and 2 tests on the latest Kali.

For sure. Could you please use a "known good" firmware which is working with Kali 2022.3.

torabi12, 2022-12-10:
I created a new Kali 2022.4 from the 7z, then
$ sudo apt update and upgrade and installer.sh -d

It dropped 2 times this:

[image]

and the CVE database download failed:

[image]

[image]

I have a snapshot of the clean system and I try again, but this is the fast experience.

torabi12, 2022-12-10:
Installation dropped only the issue [#427](https://github.com/e-m-b-a/emba/issues/427) 2 times.

I used for testing the [DVRFv0.3](https://github.com/praetorian-inc/DVRF) binary file.

Starting command for 1st test was:
$ sudo ./emba.sh -l ~/DVRF -f ~/DVRF_v03.bin -p ./scan-profiles/default-scan.emba -S -X "v0.3" -Y "DVRF" -Z "Router" -N "DVRF_v03.bin" 

and at the end:
[image]

Report looks good and the scan was also OK.
If you have any suggestions on what I should pay more attention to, please let me know. The second test will be soon.

torabi12, 2022-12-11:
So far I've run few emulation firmware tests and I only have one binary to run this with emba. I used [this]([http://www.downloads.netgear.com/files/GDC/WNAP320/WNAP320_V3.7.11.4.zip](https://www.downloads.netgear.com/files/GDC/WNAP320/WNAP320_V3.7.11.4.zip?_ga=2.129688654.316720904.1670755468-1077761920.1670755468))

Starting command:
$ sudo ./emba.sh -l ~/WNAP -f ~/WNAP320_V3.7.11.4.zip -p ./scan-profiles/default-scan-emulation.emba -S -X "V.3.7.11.4" -Y "Netgear" -Z "WNAP320" -N "WNAP_V3.7.11.4.zip" 

Ended like this:
[image]

I had only this at the beginning:
[image]

I added a not existing log folder into the command, I don't know why this question came up.

Report was also nice:
[image]

m-1-k-3, 2022-12-11:
The logs directory in the EMBA root directory is probably from a EMBA run without the log parameter `-l`. If you start EMBA without a log directory it automatically generates the default one. I have seen that also `./emba.sh -d` creates such a log directory. So, this is a bug. Thank you.

torabi12, 2022-12-11:
Yes I made a mistake, at the very beginning I entered the emba.sh command without -l -f parameters and after I executed the full command.

torabi12, 2022-12-11:
I checked the full installation mode and it stopped at this point:

[image]

and got back the prompt.

m-1-k-3, 2022-12-11:
could you please show me the output of the following command:

```
ls -ld /home/kali/emba/external/*
```

torabi12, 2022-12-11:
The folder looks like this:
[image]

m-1-k-3, 2022-12-11:
This looks correct. Currently I'm setting up another kali for further testing. First issues are already addressed with #429 and #428

m-1-k-3, 2022-12-11:
> I tried the installation 2 times and I got the same:

Got it also. I will take a look at it

torabi12, 2022-12-12:
I checked again the full installation recently and it stopped with the same error like above:

Stack Trace:
    [1] IP12_avm_freetz_ng_extract(): ./installer/IP12_avm_freetz_ng_extract.sh, line 86 -> IP12_avm_freetz_ng_extract
    [2] main(): ./installer.sh, line 259 -> main -F

torabi12, 2022-12-14:
I tried again but the installer stopped the same place:

[image]

m-1-k-3, 2022-12-14:
The full installation is not fixed yet. The docker image is used in default installation mode.

torabi12, 2022-12-14:
The default installation has been working for me for the last few days, so I thought it was worth trying the full one.
I have just installed a default one but found no difference in the process.

The red lines were there:
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
pyppeteer 1.0.1 requires pyee<9.0.0,>=8.1.0, but you have pyee 9.0.4 which is incompatible.                                                                                                                                                
ospd-openvas 22.4.0 requires packaging<21.0,>=20.4, but you have packaging 21.3 which is incompatible.                                                                                                                                     
ospd-openvas 22.4.0 requires paho-mqtt<2.0.0,>=1.6.0, but you have paho-mqtt 1.5.1 which is incompatible.

If you can help me with what to check, I'll try.

m-1-k-3, 2022-12-14:
Is on the todo list

m-1-k-3, 2022-12-16:
> I checked again the full installation recently and it stopped with the same error like above:
> 
> Stack Trace: [1] IP12_avm_freetz_ng_extract(): ./installer/IP12_avm_freetz_ng_extract.sh, line 86 -> IP12_avm_freetz_ng_extract [2] main(): ./installer.sh, line 259 -> main -F
> 
> [image]

addressed with https://github.com/e-m-b-a/emba/pull/433

torabi12, 2022-12-16:
I tried the full installation and the process stopped like before:

[image]

I used a brand new Kali 2022.4 with
$ sudo ./installer.sh -F

I cloned the emba after the #433 merge to main.

m-1-k-3, 2022-12-16:
seen it also ... we are going to check this

m-1-k-3, 2022-12-16:
update is out there ...

torabi12, 2022-12-16:
The full installation went through the previous step but it stopped here on my side:

[image]

closed 2022-12-19 by m-1-k-3