e-m-b-a/emba #427: Some incompatible package on Kali linux 2022.4

torabi12, 2022-12-10:


torabi12, 2022-12-10:
Thank you, yes I wasn not correct, but recently I wanted to use and test EMBArk.
Right now I am using Ubuntu Server 22.04.1 and I have only one finding.
I create a VM in VMWare Workstation 16, Ubunru Server 22.04.1, fully upgraded, 200GB disk space, I install EMBark and it works fine on my local system.
I create an OVA/OVF file and I move this great system to anothes server which runs ESXi 8.0 and it also works but the EMBArk web page after the successful scan does not contain the logs and reports only the file system has it.
I try again for ther 3rd time early next week, but I have o idea what could be wrong on my side.

Thanks again!

torabi12, 2022-12-10:
I installed EMBArk today and I would like to check the installation logs.
Sorry for the question but where is it?
I did not dind it in ~/embark and in /var/logs

:(

m-1-k-3, 2022-12-10:
Thank you for your testing support. Nevertheless, could we please split this thread into EMBArk issues documented [here](https://github.com/e-m-b-a/embark/issues) and EMBA issues which we are going to handle in this thread. Otherwise it will get quite confusing.

torabi12, 2022-12-17:
As we did for patch #405, I tried to fix the above bugs with these versions:

`sudo pip3 install --force-reinstall --no-deps --ignore-installed pyee==8.2.2`
`sudo pip3 install --force-reinstall --no-deps --ignore-installed packaging==20.9`
`sudo pip3 install --force-reinstall --no-deps --ignore-installed paho-mqtt==1.6.1`

I don't know if these are the best versions and if they even need to be rewritten in the docker container.

m-1-k-3, 2022-12-19:
As a quick update ... the docker container is currently fine and does not need an update.

m-1-k-3, 2022-12-19:
As a quick reference. The only conflict from the container should be solved automatiacally with the next update -https://github.com/stacscan/stacs/issues/32

torabi12, 2022-12-19:
I did a default emba installation today and colorama was fine:

[image]

I got back the red lines regarding pyppeteer and ospd-openvas requirements like above. Maybe it can be ignored, unfortunately, I can only tell if something is good if the error message disappears.
There were no other issues during the installation.

Thank you very much!

m-1-k-3, 2022-12-19:
colorama only affects the docker image, not the host.

m-1-k-3, 2022-12-20:
As far as I can see the dependency issues are from a plain Kali installation which includes openvas. 
If you do not use it you can uninstall it via `apt-get purge openvas-scanner`

m-1-k-3, 2022-12-20:
The pyee/pyppeteer dependency conflict can be automatically solved via

`sudo pip3 install -U pyppeteer`

torabi12, 2022-12-20:
I did your advices, I removed openvas and installed pyppeteer:

[image]

and started an emba installation:

[image]

If I understand correctly, emba wants to install or use the pyee, packaging and paho-mqtt packages but other applications on the latest Kali require other versions and that's why I get an error message when installing emba.

Thank you for your continued support.

closed 2022-12-22 by m-1-k-3