e-m-b-a/emba #437: CVE-Search install error

SmllXzBZ, 2022-12-19:
Hello, I encounter a problem installing cve-search.

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04.1
- EMBA version: newest
- Installation method: default with up to date docker image

step:
```
sudo ./installer.sh -d
```
then install cve-search:
```
wget -qO - https://www.mongodb.org/static/pgp/server-4.4.asc | sudo apt-key add -
echo "deb [ arch=amd64,arm64 ] https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 multiverse" | sudo tee /etc/apt/sources.list.d/mongodb-org-4.4.list
sudo apt-get update
sudo apt-get install -y mongodb-org
sudo systemctl daemon-reload
sudo systemctl start mongod
sudo ./sbin/db_mgmt_cpe_dictionary.py -p
sudo ./sbin/db_mgmt_json.py -p
sudo ./sbin/db_updater.py -c
```
but it takes a long time but no response, then I use this command:
```
$sudo ./external/cve-search/sbin/db_updater.py -f
2022-12-19 04:50:53,890 - CAPECDownloads - INFO     - Duration: 0:00:16.275875
2022-12-19 04:50:53,890 - CAPECDownloads - INFO     - Finished CAPEC database update
2022-12-19 04:50:53,891 - DBUpdater - INFO     - capec has 546 elements (546 update)
2022-12-19 04:50:53,891 - DBUpdater - INFO     - Dropping collection: via4
2022-12-19 04:50:53,897 - DBUpdater - INFO     - via4 dropped
2022-12-19 04:50:53,897 - DBUpdater - INFO     - Starting via4
2022-12-19 04:50:53,902 - VIADownloads - INFO     - VIA4 database update started
Downloading files: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [12:56<00:00, 776.69s/it]
Processing downloaded files:   0%|                                                                                                   | 0/1 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "/home/user/emba/external/cve-search/sbin/./db_updater.py", line 163, in <module>
    up.update()
  File "/home/user/emba/external/cve-search/sbin/../lib/Sources_process.py", line 713, in update
    self.process_downloads([self.feed_url], collection=self.feed_type.lower())
  File "/home/user/emba/external/cve-search/sbin/../lib/DownloadHandler.py", line 124, in process_downloads
    thread_map(
  File "/usr/local/lib/python3.10/dist-packages/tqdm/contrib/concurrent.py", line 94, in thread_map
    return _executor_map(ThreadPoolExecutor, fn, *iterables, **tqdm_kwargs)
  File "/usr/local/lib/python3.10/dist-packages/tqdm/contrib/concurrent.py", line 76, in _executor_map
    return list(tqdm_class(ex.map(fn, *iterables, **map_args), **kwargs))
  File "/usr/local/lib/python3.10/dist-packages/tqdm/std.py", line 1195, in __iter__
    for obj in iterable:
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 621, in result_iterator
    yield _result_or_cancel(fs.pop())
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 319, in _result_or_cancel
    return fut.result(timeout)
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 458, in result
    return self.__get_result()
  File "/usr/lib/python3.10/concurrent/futures/_base.py", line 403, in __get_result
    raise self._exception
  File "/usr/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/user/emba/external/cve-search/sbin/../lib/Sources_process.py", line 661, in file_to_queue
    for cve in self.ijson_handler.fetch(filename=filename, prefix=self.prefix):
  File "/home/user/emba/external/cve-search/sbin/../lib/IJSONHandler.py", line 17, in fetch
    for item in ijson.items(input_file, prefix):
ijson.common.IncompleteJSONError: parse error: premature EOF
                                       
                     (right here) ------^
```

Could you help me?

m-1-k-3, 2022-12-20:
Looks like you are using Via4CVE (https://cve-search.github.io/cve-search/database/database.html#populating-the-database). I have never used this feature. I would recommend doing a fresh installation without extra features enabled. If this is running you can play around with extra features.

m-1-k-3, 2023-01-09:
Any updates on this issue? If so, please re-open it

closed 2023-01-09 by m-1-k-3