e-m-b-a/emba #440: Various version numbers

torabi12, 2022-12-20:
**Describe the bug**

After installing emba today, the scan started with 1.1.2 and with the "-b" parameter I got 0.9.0.
If the problem is not relevant, please close and ignore it, I just wanted to ask, because I remember a few days ago I got 1.2.0 after installing.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (/default mode)
2. Use the firmware available here: any
3. Start EMBA with the following parameters: sudo ./emba.sh -l ~/logfolder -f ~/file.bin -p ./scan-profiles/default-scan.emba
4. Used emba release is not up to date

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2022.04
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Additional context**
The installation was these commands:
$ git clone https://github.com/e-m-b-a/emba.git
$ cd emba
$ sudo ./installer.sh -d

m-1-k-3, 2022-12-20:
Hi @torabi12, there should be a banner shown. It shows random banners on usual startup. Old banners are marked as *Historic Edt.*

torabi12, 2022-12-20:
Thank you. I checked some banners with -b and I found some other historic releases.
It was just a surprise that old and different versions came after a new emba installation.

closed 2022-12-20 by torabi12