e-m-b-a/emba #448: Add - CR16C support

chatelao, 2023-01-04:
**Is your feature request related to a problem? Please describe.**
I have a strange firmware for my "FritzFon" probably running on a SiTel / Dialog SC1844xx chip with a CompactRISC / CR16C instructions set. Ghidra support decompilation of the binary:
- https://download.avm.de/dect/0802/psq19p2/08.02.04.67.avm.de.upd

**Describe the solution you'd like**
If there would be a way to simulate the boot operation on the phone would be great. No idea if there are any CR16C emulators out there.

m-1-k-3, 2023-01-04:
Very interesting. Is this architecture supported from objdump or r2?

m-1-k-3, 2023-01-04:
As far as I can see on a first check EMBA is not able to extract the firmware

[image]

Freetz-NG neither

[image]

What are your steps to extract it?

m-1-k-3, 2023-01-09:
Further details on this issue?

m-1-k-3, 2023-01-31:
Please re-open it with further details.

closed 2023-01-31 by m-1-k-3