e-m-b-a/emba #449: Feature requesat FritzBox network configuration

chatelao, 2023-01-05:
This a my configuration on a FritzBox 5530, maybe this helps to boot the image futher:
https://download.avm.de/fritzbox/fritzbox-5530-fiber/deutschland/fritz.os/FRITZ.Box_5530-07.30.image

```
VANI0     Link encap:Ethernet  HWaddr 3C:37:12:63:44:ED  
          UP BROADCAST RUNNING PROMISC MULTICAST  MTU:1500  Metric:1
          RX packets:292663 errors:0 dropped:0 overruns:0 frame:0
          TX packets:861865 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:96001394 (91.5 MiB)  TX bytes:194342478 (185.3 MiB)

adsl      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:2000  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:32 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

br-omci   Link encap:Ethernet  HWaddr AA:BB:CC:AA:BB:DD  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
cpunet0   Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          inet6 addr: fd62:c5d3:4303:c662::2/64 Scope:Global
          inet6 addr: fe80::2888:b66f:bbdb:7bb3/64 Scope:Link
          UP POINTOPOINT RUNNING NOARP  MTU:8192  Metric:1
dsl       Link encap:Point-to-Point Protocol  
          inet addr:192.168.254.1  P-t-P:192.168.254.1  Mask:255.255.255.255
          UP POINTOPOINT RUNNING NOARP ALLMULTI MULTICAST  MTU:1500  Metric:1
eoam      Link encap:Ethernet  HWaddr 3E:37:12:63:44:ED  
          BROADCAST MULTICAST  MTU:1500  Metric:1
eth0      Link encap:Ethernet  HWaddr 3C:37:12:63:44:F0  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
eth1      Link encap:Ethernet  HWaddr 3C:37:12:63:44:F0  
          UP BROADCAST ALLMULTI MULTICAST  MTU:1500  Metric:1
eth2      Link encap:Ethernet  HWaddr 3C:37:12:63:44:F0  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
guest     Link encap:Ethernet  HWaddr 3C:37:12:63:44:F1  
          inet addr:192.168.179.1  Bcast:192.168.179.255  Mask:255.255.255.0
          inet6 addr: fe80::3e37:12ff:fe63:44f1/64 Scope:Link
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
guest4    Link encap:Ethernet  HWaddr 3E:37:12:63:44:F2  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
guest5    Link encap:Ethernet  HWaddr 3E:37:12:63:44:F3  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
guest_st1 Link encap:Ethernet  HWaddr 06:44:2C:C3:F3:6C  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
ifb0      Link encap:Ethernet  HWaddr EA:84:11:01:72:2B  
          BROADCAST NOARP  MTU:1500  Metric:1
ifb1      Link encap:Ethernet  HWaddr 4A:59:57:FB:4F:38  
          BROADCAST NOARP  MTU:1500  Metric:1
ing0      Link encap:Ethernet  HWaddr 02:02:02:02:02:02  
          BROADCAST MULTICAST  MTU:1500  Metric:1
ins0      Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-62-00-00-00-00-00-00-00-00  
          BROADCAST NOARP MULTICAST  MTU:1500  Metric:1
lan       Link encap:Ethernet  HWaddr 3C:37:12:63:44:F0  
          inet addr:192.168.254.1  Bcast:192.168.254.255  Mask:255.255.255.0
lan:0     Link encap:Ethernet  HWaddr 3C:37:12:63:44:F0  
          inet addr:169.254.1.1  Bcast:169.254.255.255  Mask:255.255.0.0
lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
ppptty    Link encap:UNSPEC  HWaddr 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  
          UP POINTOPOINT RUNNING NOARP MULTICAST  MTU:2000  Metric:1
rtlog0    Link encap:Ethernet  HWaddr 00:00:00:00:00:00  
          BROADCAST MULTICAST  MTU:1500  Metric:1
switch0   Link encap:Ethernet  HWaddr DA:DD:F2:48:4C:04  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
tunl0     Link encap:UNSPEC  HWaddr 00-00-00-00-38-38-00-62-00-00-00-00-00-00-00-00  
          NOARP  MTU:1480  Metric:1
vap0      Link encap:Ethernet  HWaddr 3C:37:12:63:44:F2  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
vap1      Link encap:Ethernet  HWaddr 3C:37:12:63:44:F3  
          UP BROADCAST RUNNING ALLMULTI MULTICAST  MTU:1500  Metric:1
wlan0     Link encap:Ethernet  HWaddr 06:C0:FF:EE:BA:BE  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
wlan2     Link encap:Ethernet  HWaddr 02:C0:FF:EE:BA:BE  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
```

m-1-k-3, 2023-01-12:
Thank you for these insights. But, shouldn't there be some interface with a `192.168.178.x` address?

m-1-k-3, 2023-03-30:
Did a quick check on one of my AVM firmware images.

On the one hand EMBA is able to bring the device to a reachable state:

[image]

On the other hand it looks as we have some serious issues:

[image]

Probably we need a newer kernel for this device.

m-1-k-3, 2023-03-30:
Looks as the kernel is not that new on my testfirmware:

```
┌──(m1k3㉿emba)-[~/firmware-stuff/emba_forked]
└─$ modinfo <stripped>//filesystem_core.squashfs_extract/lib/modules/3.10.107/net/ath_dfs.ko
filename:       /home/m1k3/firmware-stuff/emba_logs_fritz/firmware/unblob_extracted/firmware_extract/var/tmp/filesystem.image_extract/0-25817088.squashfs_v4_be_extract/filesystem_core.squashfs_extract/lib/modules/3.10.107/net/ath_dfs.ko
license:        Proprietary
description:    DFS Support for Atheros 802.11 wireless LAN cards.
author:         Atheros Communications, Inc.
depends:        qdf,aae,asf
vermagic:       3.10.107 SMP mod_unload MIPS_34K 32BIT 
parm:           domainoverride:int
parm:           usenol:int
                                                                                                                                                                                                             
┌──(m1k3㉿emba)-[~/firmware-stuff/emba_forked]
└─$ file <stripped>/filesystem_core.squashfs_extract/lib/modules/3.10.107/net/ath_dfs.ko
/home/m1k3/firmware-stuff/emba_logs_fritz/firmware/unblob_extracted/firmware_extract/var/tmp/filesystem.image_extract/0-25817088.squashfs_v4_be_extract/filesystem_core.squashfs_extract/lib/modules/3.10.107/net/ath_dfs.ko: ELF 32-bit MSB relocatable, MIPS, MIPS32 rel2 version 1 (SYSV), BuildID[sha1]=4c73c0000728d0eed5395bc279c75b67ee0aa16e, not stripped
```

m-1-k-3, 2023-05-23:
Recently, some progress happened with the live testing modules. These improvements also affects the AVM emulation:

[image]

Tested the following devices:
* FRITZ.Box_6850_5G-07.30.image
* FRITZ.Box_4060-07.30.image
* FRITZ.Box_6850_LTE-07.29.image
* FRITZ.Box_7530-07.29.image
* FRITZ.Box_7530_AX-07.31.image
* FRITZ.Box_7510-07.30.image
* FRITZ.Box_4040-07.29.image

Nevertheless, the web server is not fully configured which results in a non working environment.

m-1-k-3, 2023-12-01:
I think we can close this until someone will work on getting better results out of it. PR are welcome

closed 2023-12-01 by m-1-k-3