e-m-b-a/emba #459: Serial access via telnet isn't working on emulation packages

aserper, 2023-01-25:
Once a qemu full system emulation package is being produced and executed with the run.sh script, telneting to localhost on port 4321 doesn't lead anywhere. A connection is being opened but there is no output on the telnet session and input isn't accepted (the cursor doesn't move or anything). I tried that with several emulation packages generated for several different firmwares.

m-1-k-3, 2023-01-26:
If the device opens a shell the telnet access should be available. Could you share the firmware for testing?

aserper, 2023-01-27:
So far it failed on every single firmware I threw at it but I'm happy to provide several examples of firmware. Unfortunately github won't let me attach the files here is one example:
Asus - AC1300 firmware, filename: FW_RT_AC3100_300438648260 which can be downloaded from here:
https://www.asus.com/supportonly/rt-ac3100/helpdesk_bios/?model2Name=RT-AC3100 but as I said, I tried about 15 different fws from different vendors, I couldn't get a working "serial connection" via telnet on any one of them. Happy to provide more input if you need me to.

m-1-k-3, 2023-01-27:
Thank you for the link. I will give it a try.

If you want to see most of the emulation features running quite smooth you could use one of the old DIR300 firmwares here: https://ftp.dlink.de/dir/dir-300/archive/driver_software/DIR-300_fw_revb_214b01_ALL_de_20130206.zip

aserper, 2023-01-27:
I tried to debug and fix it myself but I couldn't understand where the emulated serial device gets "connected"/"mapped" to the telnet port. I tried futzing with the run.sh script but nothing I did worked on any of the 15 firmwares I was trying

m-1-k-3, 2023-01-28:
Looks like something weird is going on. If I run the qemu command manually I can use the cli interface:

```
└─$ sudo qemu-system-arm -m 2048 -M virt  -kernel ./zImage.armel -drive if=none,file=./squashfs-root_armel-31493,format=raw,id=rootfs -device virtio-blk-device,drive=rootfs -append "root=/dev/vda1 console=ttyS0 nandsim.parts=64,64,64,64,64,64,64,64,64,64 init=/firmadyne/preInit.sh rw debug ignore_loglevel print-fatal-signals=1 FIRMAE_NET=true FIRMAE_NVRAM=true FIRMAE_KERNEL=true FIRMAE_ETC=true user_debug=0 firmadyne.syscall=0" -nographic -device virtio-net-device,netdev=net0 -netdev tap,id=net0,ifname=tap684_0,script=no -audiodev driver=none,id=none
```

[image]

Starting it with the generated run script it is currently not working.

m-1-k-3, 2023-01-28:
[image]

m-1-k-3, 2023-01-28:
In general the mechanis is working: 
[image]

This screenshot is from the DIR300 firmware.

aserper, 2023-03-07:
I tried running all of the commands not through the script and still no luck

aserper, 2023-03-08:
Ok, I seemed to have found the bug. The problem is that there is some sort of weird race condition / qemu bug that doesn't let you to have the serial output going to a file AND AT THE SAME TIME having telnet access to the serial port. Removing the file output from run.sh causes the serial to telnet port work properly.

m-1-k-3, 2023-03-09:
Cool! Thanks for letting us know how to circumvent this issue. Nevertheless, I don't think we can address this in EMBA. What do you mean?

m-1-k-3, 2023-03-29:
As I don't think we can address this from EMBA side, I'll close it now.
In case you have some ideas you can re-open it for further discussions.

closed 2023-03-29 by m-1-k-3