e-m-b-a/emba #461: Status bar rendering stability

m-1-k-3, 2023-01-26:
**Describe the bug**

The EMBA status bar (activation with -B parameter) does not regenerate correctly on changing the screen size.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. Use the firmware available here: random firmware
3. Start EMBA with the following parameters: `sudo ./emba.sh -f ~/firmware-stuff/Firmware_images/testimages/web -l ~/firmware-stuff/emba_logs_test -S -p ./scan-profiles/default-scan.emba -B` -> -B parameter is important
4. Wait for EMBA startup and the status bar appears
[image]
6. Change window size with mouse to bigger and something smaller and the status bar does not regenerate correctly anymore
[image]
[image]
[image]

**Expected behavior**
Stable status bar

**Screenshots**
see above

**Desktop:**
- OS: e.g. Kali Linux 2022.03
- EMBA version: current master branch
- Installation method: default with up to date docker image

m-1-k-3, 2024-03-19:
Need to debug this in detail soon

torabi12, 2024-03-19:
Easy to reproduce:
- please start an emba scan with default-profile in terminal window size, wait a few minutes to see the bar with values and change the terminal size to full size. The status bar will be empty in the bottom for the remaining scan progress.

m-1-k-3, 2024-03-20:
In my testing szenario it is not that ugly but it looks as the calculation of the terminal size somehow breaks

Before:

[image]

After:

[image]

Fixing this will probably take a while as I'm quite busy with other things ...

torabi12, 2024-03-20:
Yes, this is the initial state, but if you wait more while the test is running the CPU, MEM graphs will fill up the screen. Or it can happen like on my side.
It is not critical, I just wanted to mention it.

closed 2024-04-17 by HoxhaEndri