e-m-b-a/emba #472: Does not run on ARM64. `exec format error`

Qwertie-, 2023-02-06:
**Describe the bug**
Attempted to run on the ARM build of Kali Linux (In a VM running on an M1 Macbook)

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
3. run `sudo docker-compose run emba`

**Expected behavior**
Expected to not crash. 

Terminal output:
```

┌──(user㉿kali)-[~/emba]
└─$ sudo docker-compose run emba                                                              
WARNING: The FIRMWARE variable is not set. Defaulting to a blank string.
WARNING: The LOG variable is not set. Defaulting to a blank string.
WARNING: The EMBA variable is not set. Defaulting to a blank string.
Creating emba_emba_run ... done
exec /bin/bash: exec format error
ERROR: 1
```

**Desktop (please complete the following information):**
- OS: Kali Linux 2022.04 (Apple Silicon ARM64 iso)
- EMBA version: Not too sure, can't run to print version. Installed today via the git clone + docker install method
- Installation method: default with up to date docker image

m-1-k-3, 2023-02-06:
Hi @Qwertie-,

Thank you for your report. EMBA is not designed to run on ARM architecture. If you can run a VM with a x64 Image you should switch. I will include this limitation to the wiki.

Best,
Mike

m-1-k-3, 2023-02-11:
Included now into the wiki: https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

Will close this issue now

closed 2023-02-11 by m-1-k-3

m-1-k-3, 2026-05-18:
If you find the issue let us know