e-m-b-a/emba #474: ./installer.sh: line 272: IF50_aggregator_common: command not found

Chinakentgao, 2023-02-06:
**Describe the bug**
xvfb
Description: Virtual Framebuffer 'fake' X server
Download-Size: 865 kB
xvfb will be newly installed.
./installer.sh: line 272: IF50_aggregator_common: command not found
------------------------------------------------------------------------------------------------
Error detected - status code 127
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer.sh, line 272
Stack Trace:
    [1] main(): ./installer.sh, line 272 -> main -l

**To Reproduce**
Steps to reproduce the behavior:
1. ./installer.sh -l

**Expected behavior**
No error reported during normal installation

**Desktop (please complete the following information):**
- OS: e.g. Kubuntu 22.10 x86_64 
- EMBA version: v1.2.1

**Additional context**
I installed it on the system/ Installer.sh - l This error will occur after 3 hours of installation.
[image]

m-1-k-3, 2023-02-06:
Thank you for reporting. The normal installation mode is `./installer.sh -d`
-l parameter should only list the dependencies. I'm wondering about 3hours of runtime. I will need to check on it.

Chinakentgao, 2023-02-07:
I found that I modified the script and added an S to make it sIF50_aggregator_common.sh, so there is no problem with the source program. Sorry to bring this issue to you

closed 2023-02-07 by Chinakentgao

m-1-k-3, 2023-02-07:
Great and thank you for your feedback.
You pointed us also to the fact that the `-l` option is deprecated and should be removed in the future. I will add a note in the installer.