e-m-b-a/emba #475: Ubuntu 22.10 lost mongodb-org-server depends on libssl1.1

Chinakentgao, 2023-02-07:
**Describe the bug**
./installer.sh -d This error occurred during installation.

**To Reproduce**
Steps to reproduce the behavior:
1. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Ubuntu 22.10
- EMBA version: v1.2.1

**Additional context**
lost mongodb-org-server depends on libssl1.1.This version of ubuntu has deleted the lower version of libssl. need according to this [issues](https://github.com/dotnet/sdk/issues/25441)

m-1-k-3, 2023-02-07:
Sorry, but we are not supporting Ubuntu 22.10 (see https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites)
But, nevertheless it should be quite easy fixing this. Currently we are checking for the availability of libssl1.1 and then install the correct version for Ubuntu 22.04. Probably it is enough to include the right download for 22.10 and we are fine.

I think we need to introduce some additinonal value to UBUNTU_OS here: https://github.com/e-m-b-a/emba/blob/12811a523b55dc8bf79b551aae451df1fa782154/installer.sh#L152
And based on this we can decide which version we are going to install in here: https://github.com/e-m-b-a/emba/blob/12811a523b55dc8bf79b551aae451df1fa782154/installer/ID1_ubuntu_os.sh#L39

If you could give it a try we are happy about a PR to support Ubuntu 22.10

m-1-k-3, 2023-02-08:
Updated the documentation here: https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

Chinakentgao, 2023-02-08:
OK thx
 

凬清沄淡
***@***.***