e-m-b-a/emba #480: libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb download returns 404

hairlessbear, 2023-02-09:
**Describe the bug**
When performing installation, `wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb -O external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb` errors out with a 404, causing installation to exit early and fail.

**To Reproduce**
Run `sudo ./installer.sh -d`

**Expected behavior**
`.deb` file is downloaded and installed successfully

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04
- EMBA version: current master branch
- Installation method: default with up to date docker image

I modified the relevant script to update the version from `18.04.20` to `18.04.21`, which seemed to fix the problem.
```diff
diff --git a/installer/ID1_ubuntu_os.sh b/installer/ID1_ubuntu_os.sh
index c3f35bb..53e2564 100755
--- a/installer/ID1_ubuntu_os.sh
+++ b/installer/ID1_ubuntu_os.sh
@@ -39,12 +39,12 @@ ID1_ubuntu_os() {
         # libssl1.1 missing
         echo -e "\\n""$BOLD""Installing libssl1.1 for mongodb!""$NC"
         # echo "deb http://security.ubuntu.com/ubuntu impish-security main" | tee /etc/apt/sources.list.d/impish-security.list
-        wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb -O external/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
-        wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb -O external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
-        dpkg -i external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
-        dpkg -i external/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
-        rm external/libssl1.1_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
-        rm external/libssl-dev_1.1.1-1ubuntu2.1~18.04.20_amd64.deb
+        wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl-dev_1.1.1-1ubuntu2.1~18.04.21_amd64.deb -O external/libssl-dev_1.1.1-1ubuntu2.1~18.04.21_amd64.deb
+        wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1-1ubuntu2.1~18.04.21_amd64.deb -O external/libssl1.1_1.1.1-1ubuntu2.1~18.04.21_amd64.deb
+        dpkg -i external/libssl1.1_1.1.1-1ubuntu2.1~18.04.21_amd64.deb
+        dpkg -i external/libssl-dev_1.1.1-1ubuntu2.1~18.04.21_amd64.deb
+        rm external/libssl1.1_1.1.1-1ubuntu2.1~18.04.21_amd64.deb
+        rm external/libssl-dev_1.1.1-1ubuntu2.1~18.04.21_amd64.deb
     fi
 
     if [[ "$WSL" -eq 1 ]]; then
```

m-1-k-3, 2023-02-10:
Thank you. Could you open a PR to fix this issue?

m-1-k-3, 2023-02-10:
As we needed the update for EMBArk I did it in PR https://github.com/e-m-b-a/emba/pull/479
Thank you for the report

BenediktMKuehne, 2023-05-04:
problem persists with ubuntu release
! needs dynamic fixing
[image]

m-1-k-3, 2023-05-06:
Is this version number part just incrementing the last area? Can we try to download versions 18.4.21 - .29 and use the one that we were able to download?

BenediktMKuehne, 2023-05-08:
> Is this version number part just incrementing the last area? Can we try to download versions 18.4.21 - .29 and use the one that we were able to download?

Appears that way

closed 2023-05-19 by m-1-k-3