e-m-b-a/emba #484: installer fail - non-free-firmware

PaulPuschmann, 2023-02-15:
with the command for installation
sudo ./installer.sh -d
using kali linux 2022.04
the following error occurs

┌──(kali㉿kali)-[~/Desktop/emba-Master]
└─$ sudo ./installer.sh -d                                                           

EMBA - Embedded Linux Analyzer Installer
=================================================================

==> Imported 25 installer module files

Install all dependecies for EMBA in default/docker mode

Created external directory: ./external

Update package lists.
Ign:1 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 InRelease
Hit:3 https://repo.mongodb.org/apt/ubuntu focal/mongodb-org/4.4 Release
Hit:2 http://ftp.halifax.rwth-aachen.de/kali kali-rolling InRelease
Traceback (most recent call last):
  File "/usr/lib/cnf-update-db", line 27, in <module>
    col.create(db)
  File "/usr/share/command-not-found/CommandNotFound/db/creator.py", line 95, in create
    self._fill_commands(con)
  File "/usr/share/command-not-found/CommandNotFound/db/creator.py", line 143, in _fill_commands
    self._parse_single_contents_file(con, f, fp.stdout)
  File "/usr/share/command-not-found/CommandNotFound/db/creator.py", line 282, in _parse_single_contents_file
    priority = component_priorities[component]
KeyError: 'non-free-firmware'
Reading package lists... Done
W: https://repo.mongodb.org/apt/ubuntu/dists/focal/mongodb-org/4.4/Release.gpg: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
E: Problem executing scripts APT::Update::Post-Invoke-Success 'if /usr/bin/test -w /var/lib/command-not-found/ -a -e /usr/lib/cnf-update-db; then /usr/lib/cnf-update-db > /dev/null; fi'
E: Sub-process returned an error code
------------------------------------------------------------------------------------------------
Error detected - status code 100
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer.sh, line 228
Stack Trace:
    [1] main(): ./installer.sh, line 228 -> main -d

m-1-k-3, 2023-02-15:
Interesting. Line 228 is doing an `apt-get update`
Could you do a manual update of your environment on the command line:

```
sudo apt-get update
sudo apt-get upgrade
```

PaulPuschmann, 2023-02-16:
i solved the problem
My network configuration was the problem

closed 2023-02-16 by PaulPuschmann

m-1-k-3, 2023-02-16:
good to hear