e-m-b-a/emba #486: Check system memory and warn if < 4GB

eacmen, 2023-02-17:
**Is your feature request related to a problem? Please describe.**
I was trying to use emba and the cve_search module always threw an error. Deeper inspection revealed a ton of Redis exceptions (connection reset, connection refused) in the db_updater.py code. Some hours of debugging later on revealed that was hitting an upper limit as the kali VM I imported only had 2GB of RAM by default. 

Sorry I don't have copies of the exact exceptions I was getting.

**Describe the solution you'd like**
When installing or running cve-search module recommend checking system memory and issuing a warning if the system has less than 4GB of RAM.

**Describe alternatives you've considered**
none

**Additional context**
Great tool. Thanks!

m-1-k-3, 2023-02-17:
Thats a very good idea. As we already have a warning if there is not enough disk space such an additional check totally makes sense

m-1-k-3, 2023-02-17:
Did a quick fix in the next PR here https://github.com/e-m-b-a/emba/pull/485

Thanks for your report.

eacmen, 2023-02-17:
wow that was quick. thank you! ill test it out when i get a chance.

closed 2023-02-23 by BenediktMKuehne