e-m-b-a/emba #488: can not update and install

Roksolana969, 2023-02-23:
┌──(root㉿kali)-[/home/kali/Desktop/emba]
└─# ./installer.sh -h
./helpers/helpers_emba_load_strict_settings.sh: line 24: DEBUG_SCRIPT: unbound variable
                                                                                                                                                                                                                 
┌──(root㉿kali)-[/home/kali/Desktop/emba]
└─# ./installer.sh -d
./helpers/helpers_emba_load_strict_settings.sh: line 24: DEBUG_SCRIPT: unbound variable
                                                                                                                                                                                                                 
┌──(root㉿kali)-[/home/kali/Desktop/emba]
└─#

Roksolana969, 2023-02-23:
ohno. i  solved,  i change the script 1 to 0

closed 2023-02-23 by Roksolana969

m-1-k-3, 2023-02-23:
Thank you for using EMBA and great that you solved it ;)