e-m-b-a/emba #489: Error mounting "/etc/timezone" to rootfs at "/etc/timezone"

themysterian-sec, 2023-02-23:
Hi,
I would like to test EMBA for the first time. I had to comment the "$DEBUG_SCRIPT" line in  helpers/helpers_emba_load_strict_settings.sh 
as it won't install properly (./helpers/helpers_emba_load_strict_settings.sh: line 24: DEBUG_SCRIPT: unbound variable
). But Setup finishes then without errors.
Installation on my Kali Linux up-to-date version 2022.4. It is a Intel 64Bit Laptop.

When starting emba with
./emba -l ./log -f /home/pentest/Downloads/Hirschmann/m100geL2P.bin
I receive an error and the process stops.

Error Message at the bottom:

[+] Dependency check
=================================================================

[*] Elementary:
    user permissions
    user permission - ok
    configuration directory - ok

[*] Necessary utils on system:
    docker - ok
    docker-compose - ok
    MongoDB - ok
    Docker Interface - ok
    cve-search - testing
    cve-search - ok
    inotifywait - ok
    notify-send - ok

[*] Started kernel downloader thread with PID 13234
[*] Original user: pentest
[*] Notification process started with PID 13248
[*] Firmware binary detected.
    EMBA starts with the pre-testing phase.
    EMBA is running with 1 modules in parallel and 1 threads per module.

    cve-search - testing
[*] EMBA sets up the docker environment.

[*] EMBA initializes docker container.

Creating emba_emba_run ... done
Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/etc/timezone" to rootfs at "/etc/timezone": mount /etc/timezone:/etc/timezone (via /proc/self/fd/7), flags: 0x5001: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
ERROR: 1
[-] EMBA failed in docker mode!
[*] Final cleanup started.
[*] Stopping kernel downloader thread with PID 13234
[*] Stopping EMBA PID 13248

m-1-k-3, 2023-02-23:
The issue around the DEBUG_SCRIPT will be solved in #485

m-1-k-3, 2023-02-23:
What is your output of the following?

```
└─$ file /etc/timezone
/etc/timezone: ASCII text
```

For testing you could remove the entry from docker-compose.yml
```
        - /etc/timezone:/etc/timezone:ro
```

themysterian-sec, 2023-02-24:
The output is

> $ file /etc/timezone
> /etc/timezone: directory

Directory itself is empty
>└─$ ll /etc/timezone
> total 0

In the settings it is also empty
[image]

Commenting the line does the trick.

closed 2023-02-24 by m-1-k-3