e-m-b-a/emba #507: fork: retry: Resource temporarily unavailable

PaulPuschmann, 2023-03-02:
**Describe the bug**
When I start emba everything runs fine until S115_usermode_emulator.sh is called in the analysis, then the following messages appear. 
./modules/S115_usermode_emulator.sh: fork: retry: Resource temporarily unavailable
./helpers/helpers_emba_print.sh: fork: retry: Resource temporarily unavailable

these errors keep repeating themselves until the end of the analyze phase

The program usually runs through to the end, but when creating the output, emba notes again that there is no data for S115

I can assure you that the error is not due to a lack of performance or a set limit on the processes: see ulimit -a and limits.conf

**To Reproduce**
starting emba with different commands
sudo ./emba -l ./rootfs -f /home/vulnscan/Schwachstellenscanner/rootfs -p ./scan-profiles/default-scan.emba
sudo ./emba -l ./rootfs -f /home/vulnscan/Schwachstellenscanner/rootfs -S -p ./scan-profiles/default-scan.emba -m s20 -m s115 -m f199 -y -j

**Expected behavior**
complete the full analysis with output from S115

**Screenshots**
Output in Analyse Phase
[image]

ulimit -a:
[image]

etc/security/limits.conf:
[image]

**Desktop**
- OS: Ubuntu jammy 22.04.2 LTS
- EMBA current master branch
- Installation method: default with up to date docker image
- Enough computing power

m-1-k-3, 2023-03-02:
Could you provide a testing firmware for this issue

PaulPuschmann, 2023-03-02:
unfortunately I cannot share the firmware with you. :(
But it's actually a rootfs, of a linux system

m-1-k-3, 2023-03-02:
Is the error only with this firmware or also with others? Please find a firmware you can share for reproducing.

In the mean time you can try to tear it down a bit:
* run `dmesg` while testing
* run only s115 with `-E -m s115 -m f199`
* run in unthreaded mode with something like ./emba -f ~/firmware -l ~/logs -S -z -s -E -m s115 -m f199
* Test it on Kali linux to check if it is an Ubuntu only issue or also happens on Kali Linux

PaulPuschmann, 2023-03-02:
I'll see if I can find a similar firmware

Here you see the terminal if i run:

sudo ./emba -l ./rootfs -f /home/vulnscan/Schwachstellenscanner/rootfs -S -p ./scan-profiles/default-scan.emba **-m s115 -m f199**

In the text file you will only find the analysis phase, as the preparation was unremarkable

The problems start at line 2331

[emba output.txt](https://github.com/e-m-b-a/emba/files/10869976/emba.output.txt)

closed 2023-03-06 by PaulPuschmann