e-m-b-a/emba #508: pip packages on latest Kali

torabi12, 2023-03-03:
**Describe the bug**
I got some errors on the latest Kali for the following packages:
[image]

[image]

[image]

I wanted to fix them by manually, but it was not working:

[image]

If the issu is not relevant please ignore it, I just wanted to collect and mention what is happening on a new installation with a brnad new Kali linux.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. See errors

**Expected behavior**
If it can be fixed please give me some advice.

**Screenshots**
See above.

**Desktop (please complete the following information):**
- OS: Kali Linux 2022.4
- EMBA version: current master branch
- Installation method: default with up to date docker image

m-1-k-3, 2023-03-03:
You should be able to install packages with the additional parameter `--break-system-packages`
See also here: https://github.com/e-m-b-a/emba/blob/2285e2aedeb37138455bca30df3a74b7d3ebd040/installer/helpers.sh#L174

torabi12, 2023-03-07:
I tried these packages and I didn't get the warnings during the new emba installation:

$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed aioconsole==0.3.3
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed lsassy==3.1.4
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed minikerberos==0.3.3
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed neo4j==4.4.10
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed pypsrp==0.7.0
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed pywerview==0.3.3
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed xmltodict==0.12.0
$ sudo pip3 install --force-reinstall --break-system-packages --no-deps --ignore-installed pyee==8.2.2

closed 2023-03-10 by torabi12

m-1-k-3, 2023-04-30:
see https://github.com/e-m-b-a/emba/issues/568