e-m-b-a/emba #519: I can't use the EMBA in Docker.

dragosmanolea, 2023-03-10:
**Describe the bug**
Hello! I cloned the the EMBA repository and I cloned run the docker-compose run command, as mentioned in the Wiki section.
The build is ok, but when i run the emba binary, it fails when it checks the necessary utils on system:
Wrong mongodb config - check your installation
RE-run instalation - bindIp should be set on 172.36.0.1
... etc (more details in Screenshot) 

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repo
2. Run docker-compose build
3. Run the docker image like in Wiki example
4. Run the emba binary: emba -f /firmware -l /log
5. See the errors

**Expected behavior**
I am expected to run the binary because i use a docker container.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Ubuntu Server 20 VM
- EMBA version: v1.2.1
- Installation method: default with up to date docker image

**Additional context**
Add any other context about the problem here.

m-1-k-3, 2023-03-10:
You need to run the installer command before using EMBA:

```
cd EMBA-directory
sudo ./installer.sh -d
```

dragosmanolea, 2023-03-10:
When i try to install it, i have installation problems.

dragosmanolea, 2023-03-10:
I mention that i used the docker image from dockerhub and had the same issue.

m-1-k-3, 2023-03-10:
your installation is lacking multiple components. Checked the https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites ?

dragosmanolea, 2023-03-10:
I will try to build the docker image on Ubuntu:jammy and i will come with updates.
Thank

m-1-k-3, 2023-03-10:
I recommend using the latest master branch of EMBA and the docker image which is used by the installer for your first try.

m-1-k-3, 2023-03-10:
You can find the installation instructions here: https://github.com/e-m-b-a/emba#installation

m-1-k-3, 2023-03-10:
EMBA has a pre-build docker image for the main analysis. Additionally there are some components needed outside the docker container. If you are going to build the docker container by yourself this is fine but you need the outside components installed somehow. You can do this by running the installer in `-d` mode first, then build your own container and you should be good to go. Nevertheless, you could also just run installer.sh -d and everything should be ready to run afterwards ;)

dragosmanolea, 2023-03-10:
I don't want necessarily to build it again. I want to run the binary, but i'm confused from the documentation what i am doing wrong.

dragosmanolea, 2023-03-10:
I followed this steps on a fresh Ubuntu VM 22.04:
1. Cloned the latest version from the repo
2. Inside emba directory, run: docker-compose build
3. The build was successfully
4. Run the docker: EMBA="." FIRMWARE=/absolute/path/to/firmware LOG=/home/n/firmware_log/ docker-compose run emba, with the correct path.

Results:
[image]

What am i doing wrong?
Thanks

m-1-k-3, 2023-03-10:
you need to start EMBA without any docker-compose command ... just do something like this from the EMBA directory on your host:

```
sudo ./emba -f ~/PATH_TO_YOUR_FIRMWARE -l ~/log-dir -p ./scan-profiles/default-scan.emba
```

dragosmanolea, 2023-03-10:
i want to use it in a docker container

dragosmanolea, 2023-03-10:
It works from host.
On 28th feb 2023 i tried to run it in a docker container and it worked.
I pulled the latest docker image from that time and then, using docker-compose run, everything went fine.
How can i get that version of docker image?

dragosmanolea, 2023-03-14:
i managed to run everything inside one container.

closed 2023-03-14 by dragosmanolea