e-m-b-a/emba #520: Docker-compose error

dragosmanolea, 2023-03-10:
**Describe the bug**
I tried to run the docker-compose build command and i had an error during building: Job for redis-server.service failed because a timeout was exceeded.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA build docker-compose
2. Fail at installation

**Expected behavior**
Build image successfully

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Ubuntu Server 20 VM
- EMBA version: v1.2.1
- Installation method: default with up to date docker image

m-1-k-3, 2023-03-10:
For using EMBA I recommend to use the provided docker images. This is automatically installed by the installer via `sudo ./installer.sh -d`

For building the EMBA image from scratch you need to run `sudo docker-compose build --no-cache --pull` on a current Kali or Ubuntu 22.04 LTS

dragosmanolea, 2023-03-10:
When i tried to use the provided docker images, i had some problems during runtime. I opened an issue before.

m-1-k-3, 2023-03-10:
I think this issue is handled via https://github.com/e-m-b-a/emba/issues/519

closed 2023-03-10 by m-1-k-3