e-m-b-a/emba #521: Docker-compose.yml is invalid

dragosmanolea, 2023-03-10:
**Describe the bug**
When I tried to build the docker-compose i got this error: The compose file './docker-compose.yml' is invalid because:
networks.emba_runs value Additional properties are not allowed ('name' was unexpected)

**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repo
2. Run docker-compose build

**Expected behavior**
Build should work.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Ubuntu Server 20 VM
- EMBA version: v1.2.1
- Installation method: default with up to date docker image

m-1-k-3, 2023-03-10:
Are you using Ubuntu 22.04 LTS as host? Please check the https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites

dragosmanolea, 2023-03-10:
No, I have another host.
My fault here.

m-1-k-3, 2023-03-10:
I think this issue is handled via https://github.com/e-m-b-a/emba/issues/519

closed 2023-03-10 by m-1-k-3