e-m-b-a/emba #544: ubuntu22.04 UI install handle man-db (2.10.2-1) errors and mongod service start error

Chinakentgao, 2023-03-24:
**Describe the bug**
I encountered a problem similar to [https://github.com/e-m-b-a/emba/issues/301](issues-301) when installing emba on ubuntu22.04 UI
正在设置 mongodb-org-shell (4.4.19) ...
正在设置 mongodb-database-tools (100.7.0) ...
正在设置 mongodb-org-mongos (4.4.19) ...
正在设置 mongodb-org-database-tools-extra (4.4.19) ...
正在设置 mongodb-org-tools (4.4.19) ...
正在设置 mongodb-org (4.4.19) ...
正在处理用于 man-db (2.10.2-1) 的触发器 ...
Created symlink /etc/systemd/system/multi-user.target.wants/mongod.service → /lib/systemd/system/mongod.service.

The cve-search database will be downloaded and updated!

Check if the cve-search database is already installed and populated.

cve-search database not ready.

The installer is going to populate the database.
Restarting redis-server (via systemctl): redis-server.service.
Traceback (most recent call last):
  File "/home/ubuntu/Desktop/emba/external/cve-search/./sbin/db_mgmt_cpe_dictionary.py", line 27, in <module>
    from lib.Sources_process import CPEDownloads
  File "/home/ubuntu/Desktop/emba/external/cve-search/sbin/../lib/Sources_process.py", line 17, in <module>
    from lib.DatabaseLayer import (
  File "/home/ubuntu/Desktop/emba/external/cve-search/sbin/../lib/DatabaseLayer.py", line 35, in <module>
    mongo_version = db.command("buildinfo")["versionArray"]
  File "/usr/local/lib/python3.10/dist-packages/pymongo/database.py", line 757, in command
    with self.__client._socket_for_reads(
  File "/usr/lib/python3.10/contextlib.py", line 135, in __enter__
    return next(self.gen)
  File "/usr/local/lib/python3.10/dist-packages/pymongo/mongo_client.py", line 1387, in _socket_for_reads
    server = self._select_server(read_preference, session)
  File "/usr/local/lib/python3.10/dist-packages/pymongo/mongo_client.py", line 1346, in _select_server
    server = topology.select_server(server_selector)
  File "/usr/local/lib/python3.10/dist-packages/pymongo/topology.py", line 244, in select_server
    return random.choice(self.select_servers(selector,
  File "/usr/local/lib/python3.10/dist-packages/pymongo/topology.py", line 202, in select_servers
    server_descriptions = self._select_servers_loop(
  File "/usr/local/lib/python3.10/dist-packages/pymongo/topology.py", line 218, in _select_servers_loop
    raise ServerSelectionTimeoutError(
pymongo.errors.ServerSelectionTimeoutError: 172.36.0.1:27017: [Errno 111] Connection refused, Timeout: 30s, Topology Description: <TopologyDescription id: 641d20ad9478a9f027738e36, topology_type: Single, servers: [<ServerDescription ('172.36.0.1', 27017) server_type: Unknown, rtt: None, error=AutoReconnect('172.36.0.1:27017: [Errno 111] Connection refused')>]>

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: <Insert URL>
3. Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>
4. additional steps
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: ubuntu22.04 UI
- EMBA version: v1.2.2
- Installation method: sudo ./install.sh -d

**Priority issue**
- Y

**Additional context**
ubuntu@ubuntu-virtual-machine:~/Desktop/emba$ service mongod status
× mongod.service - MongoDB Database Server
     Loaded: loaded (/lib/systemd/system/mongod.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Fri 2023-03-24 14:42:17 CST; 1s ago
       Docs: https://docs.mongodb.org/manual
    Process: 19216 ExecStart=/usr/bin/mongod --config /etc/mongod.conf (code=exited, status=14)
   Main PID: 19216 (code=exited, status=14)
        CPU: 495ms

3月 24 14:42:16 ubuntu-virtual-machine systemd[1]: Started MongoDB Database Server.
3月 24 14:42:17 ubuntu-virtual-machine systemd[1]: mongod.service: Main process exited, code=exited, status=14/n/a
3月 24 14:42:17 ubuntu-virtual-machine systemd[1]: mongod.service: Failed with result 'exit-code'.

m-1-k-3, 2023-03-24:
Hi,

priority issues are for EMBA sponsors only. I have adopted it.

Have you tried to follow the provided inststructions in https://github.com/e-m-b-a/emba/issues/301 and https://github.com/e-m-b-a/emba/issues/187

Additionally, I had some issues last time I installed EMBA with low RAM. There I needed to upgrade RAM to more than 8gigs

Chinakentgao, 2023-03-24:
> Hi,
> 
> priority issues are for EMBA sponsors only. I have adopted it.
> 
> Have you tried to follow the provided inststructions in #301 and #187
> 
> Additionally, I had some issues last time I installed EMBA with low RAM. There I needed to upgrade RAM to more than 8gigs

OK I try to upgrade RAM to more than 8gigs and install again

Chinakentgao, 2023-03-24:
@m-1-k-3 I'm sorry, I tried the methods in the two links you gave, and the final result is that the mongodb service cannot be started

m-1-k-3, 2023-03-24:
looks like the mongodb is not starting up correct. Something corrupted something around mongodb ;)

I recommend walking through the following debugging steps:
* upgrade to around 10+ Gig of RAM
* after installing EMBA (and probably failing CVE-search install) reboot the machine
* after reboot do a manual restart of the complete cve-search environment (mongodb, redis)
* check the services via netstat -> if 27017 is up you can try to populate it manually with external/cve-search/sbin/db_updater.py
* if this fails we need to go into detailed cve-search debugging and probably re-install cve-search manually and check what step is failing (see here https://cve-search.github.io/cve-search/)

btw in the meantime you can use EMBA also without cve-search -> ./emba -j should help you for this

m-1-k-3, 2023-03-24:
and there is a mongodb log in /var/log/mongodb/mongod.log. Check this for errors. Most of the time mongodb screams loud and google is helping with the error messages.

Chinakentgao, 2023-03-24:
> and there is a mongodb log in /var/log/mongodb/mongod.log. Check this for errors. Most of the time mongodb screams loud and google is helping with the error messages.

@m-1-k-3 I'm sorry I looked at the log and couldn't find out how to solve the error problem
The following is the error message of the /var/log/mongodb/mongod.log

{"t":{"$date":"2023-03-24T15:40:07.022+08:00"},"s":"I",  "c":"CONTROL",  "id":23285,   "ctx":"main","msg":"Automatically disabling TLS 1.0, to force-enable TLS 1.0 specify --sslDisabledProtocols 'none'"}
{"t":{"$date":"2023-03-24T15:40:07.026+08:00"},"s":"I",  "c":"NETWORK",  "id":4648601, "ctx":"main","msg":"Implicit TCP FastOpen unavailable. If TCP FastOpen is required, set tcpFastOpenServer, tcpFastOpenClient, and tcpFastOpenQueueSize."}
{"t":{"$date":"2023-03-24T15:40:07.026+08:00"},"s":"I",  "c":"STORAGE",  "id":4615611, "ctx":"initandlisten","msg":"MongoDB starting","attr":{"pid":3412,"port":27017,"dbPath":"/var/lib/mongodb","architecture":"64-bit","host":"ubuntu-virtual-machine"}}
{"t":{"$date":"2023-03-24T15:40:07.026+08:00"},"s":"I",  "c":"CONTROL",  "id":23403,   "ctx":"initandlisten","msg":"Build Info","attr":{"buildInfo":{"version":"4.4.19","gitVersion":"9a996e0ad993148b9650dc402e6d3b1804ad3b8a","openSSLVersion":"OpenSSL 1.1.1f  31 Mar 2020","modules":[],"allocator":"tcmalloc","environment":{"distmod":"ubuntu2004","distarch":"x86_64","target_arch":"x86_64"}}}}
{"t":{"$date":"2023-03-24T15:40:07.026+08:00"},"s":"I",  "c":"CONTROL",  "id":51765,   "ctx":"initandlisten","msg":"Operating System","attr":{"os":{"name":"Ubuntu","version":"22.04"}}}
{"t":{"$date":"2023-03-24T15:40:07.026+08:00"},"s":"I",  "c":"CONTROL",  "id":21951,   "ctx":"initandlisten","msg":"Options set by command line","attr":{"options":{"config":"/etc/mongod.conf","net":{"bindIp":"172.36.0.1","port":27017},"processManagement":{"timeZoneInfo":"/usr/share/zoneinfo"},"storage":{"dbPath":"/var/lib/mongodb","journal":{"enabled":true}},"systemLog":{"destination":"file","logAppend":true,"path":"/var/log/mongodb/mongod.log"}}}}
{"t":{"$date":"2023-03-24T15:40:07.026+08:00"},"s":"E",  "c":"STORAGE",  "id":20557,   "ctx":"initandlisten","msg":"DBException in initAndListen, terminating","attr":{"error":"IllegalOperation: Attempted to create a lock file on a read-only directory: /var/lib/mongodb"}}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"REPL",     "id":4784900, "ctx":"initandlisten","msg":"Stepping down the ReplicationCoordinator for shutdown","attr":{"waitTimeMillis":10000}}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"COMMAND",  "id":4784901, "ctx":"initandlisten","msg":"Shutting down the MirrorMaestro"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"SHARDING", "id":4784902, "ctx":"initandlisten","msg":"Shutting down the WaitForMajorityService"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"NETWORK",  "id":20562,   "ctx":"initandlisten","msg":"Shutdown: going to close listening sockets"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"NETWORK",  "id":4784905, "ctx":"initandlisten","msg":"Shutting down the global connection pool"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"STORAGE",  "id":4784906, "ctx":"initandlisten","msg":"Shutting down the FlowControlTicketholder"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"-",        "id":20520,   "ctx":"initandlisten","msg":"Stopping further Flow Control ticket acquisitions."}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"NETWORK",  "id":4784918, "ctx":"initandlisten","msg":"Shutting down the ReplicaSetMonitor"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"SHARDING", "id":4784921, "ctx":"initandlisten","msg":"Shutting down the MigrationUtilExecutor"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"CONTROL",  "id":4784925, "ctx":"initandlisten","msg":"Shutting down free monitoring"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"STORAGE",  "id":4784927, "ctx":"initandlisten","msg":"Shutting down the HealthLog"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"STORAGE",  "id":4784929, "ctx":"initandlisten","msg":"Acquiring the global lock for shutdown"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"-",        "id":4784931, "ctx":"initandlisten","msg":"Dropping the scope cache for shutdown"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"FTDC",     "id":4784926, "ctx":"initandlisten","msg":"Shutting down full-time data capture"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"CONTROL",  "id":20565,   "ctx":"initandlisten","msg":"Now exiting"}
{"t":{"$date":"2023-03-24T15:40:07.027+08:00"},"s":"I",  "c":"CONTROL",  "id":23138,   "ctx":"initandlisten","msg":"Shutting down","attr":{"exitCode":100}}

m-1-k-3, 2023-03-24:
This one? https://techoverflow.net/2019/07/22/how-to-fix-mongodb-exception-in-initandlisten-illegaloperation-attempted-to-create-a-lock-file-on-a-read-only-directory-var-lib-mongodb-terminating/

Chinakentgao, 2023-03-24:
@m-1-k-3 Thx run successfully

ubuntu@ubuntu-virtual-machine:~/Desktop$ sudo systemctl status mongod
● mongod.service - MongoDB Database Server
     Loaded: loaded (/lib/systemd/system/mongod.service; enabled; vendor preset>
     Active: active (running) since Fri 2023-03-24 16:09:26 CST; 8s ago
       Docs: https://docs.mongodb.org/manual
   Main PID: 335815 (mongod)
     Memory: 162.5M
        CPU: 521ms
     CGroup: /system.slice/mongod.service
             └─335815 /usr/bin/mongod --config /etc/mongod.conf

3月 24 16:09:26 ubuntu-virtual-machine systemd[1]: Started MongoDB Database Ser>

closed 2023-03-24 by Chinakentgao

m-1-k-3, 2023-03-24:
I have added a note to our main cve-search issue solution collection here https://github.com/e-m-b-a/emba/issues/187

Chinakentgao, 2023-03-24:
It's my pleasure :)
 

凬清沄淡
***@***.***