e-m-b-a/emba #568: crackmapexec python package error

BenediktMKuehne, 2023-04-04:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default apps_host

**Expected behavior**
installation without errors or warnings

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.01
- EMBA version: current master branch
- Installation method: default

**Additional context**

[Sponsor]: https://github.com/sponsors/e-m-b-a

torabi12, 2023-04-04:
Similar dependencies have been mentioned here:
https://github.com/e-m-b-a/emba/issues/508

m-1-k-3, 2023-04-04:
Does this affect EMBA beside the error message?

torabi12, 2023-04-04:
Anyone who doesn't know exactly how these work and installs EMBA will be scared to death that there is a bug in the system:)

m-1-k-3, 2023-04-04:
harhar ... if there is no real defect I can live with this ;)

m-1-k-3, 2023-04-30:
If you do not need [crackmapexec](https://wiki.porchetta.industries), e.g., if you are using your Kali box only for _EMBA_ you should be save to uninstall `crackmapexec` the following way:

```
   └─$ sudo apt-get remove crackmapexec -y
```

btw. this is the default VM from kali:

[image]

Which means that this is definitive not an EMBA issue. From EMBA side this is closed but I will leave this issue open for reference and that others will find it easily.

closed 2023-05-19 by m-1-k-3