e-m-b-a/emba #577: EMBA failing in docker mode

keeplearning-rgb, 2023-04-10:
I am facing this issue while trying to scan the files using EMBA.

I tried re installing EMBA several times but still facing this issue.

Please have a look at this and let me know how to fix it.

[image]

m-1-k-3, 2023-04-10:
I don't get this screenshot. The latest docker image has all these prerequisites included. Could you please give more details on your steps to get this output?

keeplearning-rgb, 2023-04-10:
This is after installing latest version of EMBA. while trying to run it on a bin file. I am getting this output

sudo ./emba.sh -f /root/Desktop/P3225-LVE_Mk_II_9_80_3_13.bin -l ~/Documents/projects/p3225_log/ -p ./scan-profiles/default-scan-emulation.emba

This is the command i have used

m-1-k-3, 2023-04-10:
a ... yes, now I see your issue. You are using the EMBArk version? Please update the EMBA installation as documented here: https://github.com/e-m-b-a/embark/issues/110#issuecomment-1500885694

keeplearning-rgb, 2023-04-10:
You want me to replace the file in helpers module and re run the installation

m-1-k-3, 2023-04-10:
you can do a `git submodule update --remote` or you can change the helper script manually

keeplearning-rgb, 2023-04-10:
Running git command didn't worked for me but updating the helpers module worked.
Thanks for the quick help.
Can't this be fixed at root level in EMBA repository so that we don't get this error while using installer.sh .

Also, if you can let me how can i install a specific version of EMBA without this version mismatch.

m-1-k-3, 2023-04-10:
We are going to fix this the next days also in EMBArk. If you just want to run EMBA without EMBArk you can install EMBA manually in a separate directory without the mismatch.

keeplearning-rgb, 2023-04-10:
Thanks for the update
I think i am still facing some issues with the same. I will update you here once it is working properly

keeplearning-rgb, 2023-04-10:
I am separately using EMBA for now. its working fine. 
I will delete my embark folder and see the fix you have provided works or not.
Please close the issue for now if required i will reopen it

closed 2023-04-11 by m-1-k-3