e-m-b-a/emba #600: TypeError: cannot unpack non-iterable yara.StringMatch object

hnguy185, 2023-04-27:
**Describe the bug**
Encountered `TypeError: cannot unpack non-iterable yara.StringMatch object` at `static anaysis of firmware for password hashes` step. Then, the scan went to, assuming, infinite loop at `check for code pattern with yara` and had to be interrupted. 

Please see the screen shot for more detail.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation: default mode
2. Use the firmware available here: https://kb.netgear.com/000060419/WNAP320-Firmware-Version-3-7-11-4
3. Start EMBA with the following parameters: sudo ./emba.sh -f ~/Downloads/WNAP320_V3.7.11.4.zip -l ~/Downloads/logs/-p ./scan-profiles/default-scan.emba 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**

[image]

**Desktop (please complete the following information):**
- OS: e.g. Ubuntu jammy 22.04
- EMBA version: current master branch
- Installation method: default with up to date docker image

m-1-k-3, 2023-04-27:
Thank you for reporting. I will try to reproduce it and will come back

m-1-k-3, 2023-04-27:
I can reproduce this issue with a default EMBA installation.

CaseGuide, 2023-04-28:
Also receive this error. Some details: installed today (04/28/2023) running on WSL2 Kali with default-scan.emba and a default install.

m-1-k-3, 2023-04-28:
Looks as the STACS installation in the current EMBA container is broken.

m-1-k-3, 2023-04-29:
As the installation on the latest Kali is not fully working I opened an issue at stacs here https://github.com/stacscan/stacs/issues/35.

m-1-k-3, 2023-05-03:
The installation issues are fixed with this PR https://github.com/e-m-b-a/emba/pull/608
The error is currently not fixed. Looks as there is something broken in the EMBA installation process.