e-m-b-a/emba #618: Scan stuck at 'Check for code patterns with yara'

hnguy185, 2023-05-10:
**Describe the bug**
This is a follow up to issue #600. After the docker was updated to the latest build, yara type error mentioned in issue #600 was resolved. However, the scan was stuck at 'Check for code patterns with yara' which is the next part of the scan and had to be interrupted. 

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: https://kb.netgear.com/000060419/WNAP320-Firmware-Version-3-7-11-4
3. Start EMBA with the following parameters: sudo ./emba.sh -f ~/Downloads/WNAP320_V3.7.11.4.zip -l ~/Downloads/logs/-p ./scan-profiles/default-scan.emba
4. additional steps
5. See error

**Desktop (please complete the following information):**
- OS: Ubuntu 22.04
- EMBA version: v1.2.2 or current master branch
- Installation method: default with up to date docker image

m-1-k-3, 2023-05-10:
hmmm. I am totally confused as the Yara module is disabled in the default profile here https://github.com/e-m-b-a/emba/blob/86a62b2aed7d0e9d0aa828af44db2fd0df8287aa/scan-profiles/default-scan.emba#L22

m-1-k-3, 2023-05-10:
Did a quick test with the default profile and the module s110 is not running:

```
[*] Wed May 10 10:03:37 CEST 2023 - S110_yara_check not executed - blacklist triggered 
```