e-m-b-a/emba #645: Kali Linux 2023.2 support

m-1-k-3, 2023-05-31:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2023-2-release/

We need to test EMBA on it

**Testcases:**

- [] Default/docker installation working with current docker image (`./installer.sh -d`)
  - [] dependency check (`./emba.sh -d`)
  - [] EMBA run with profile default-scan in strict mode (-S)
  - [] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [] EMBA run with profile full-scan in strict mode (-S)
- [] Full installation working (`./installer.sh -F`)
  - [] dependency check (`./emba.sh -d`)
  - [] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q
- [] Docker build (`sudo docker-compose build --no-cache --pull`)
  - [] dependency check (`./emba.sh -d`)
  - [] EMBA run with profile default-scan in strict mode (-S)
  - [] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [] EMBA run with profile full-scan in strict mode (-S)

torabi12, 2023-06-01:
Test env: VMWare Workstation Pro 17.0.2 & Kali 2023.2

Before emba installation I removed the
```
$ sudo apt purge openvas-scanner
$ sudo apt purge crackmapexec
```
packages. 

1st installation process was OK, CVE data came withput error, I only get some warning about pip packages:
[image]
Maybe these versions would be OK:

```
pyee==8.2.2
impacket==0.9.24
rich==10.16.2
```

- default-scan in strict mode was this:
`./emba -l /home/seclab/Kankun -f /home/seclab/kkeps.bin -S -p ./scan-profiles/default-scan.emba -X 1.0 -Y Kankun -Z Smartplug -N kkeps.bin`

I used a custom jtr_wordlist.txt file and john has also cracked the found hashes, report was OK.

- default-scan-emulation in strict mode is running right now but this failure comes back all the time:
[image]

I am using a WNAP320 fw which is easy to use with` ./fat.py` to get the dynamic analysis. I let it run for a while but it looks like the process is hanging at the above error.
The fw WNAP320_V2.0.3.zip can be found here:
https://www.netgear.com/support/product/wnap320#download

m-1-k-3, 2023-06-01:
The error looks like an error from the original S045_lighttpd.sh script in line 40. This should not stop EMBA. I will give it a try.

m-1-k-3, 2023-06-01:
we changed the startup handling of services in the last updates. Witch this we are able to cover more startup services. So, this error could be quite new :)

m-1-k-3, 2023-06-04:
The S120 module is currently not working in full-install aka dev mode. I will open a dedicated issue for this. But, as EMBA should be used in docker/default mode this issue is not critical.

Thanks for testing efforts @torabi12

closed 2023-06-04 by m-1-k-3