e-m-b-a/emba #666: Mongo Databse error

wwebest2405, 2023-06-19:
I recently decide to turn my computer in to a dual boot between ubuntu 22.04 and Windows 10 but when i tried to run sudo ./installer.sh -d in the terminal after cloning the repo I run into some Mongo DB error.

**The Error:**
Installing libssl1.1 for mongodb!
--2023-06-19 10:45:39--  http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl-dev_1.1.1-1ubuntu2.1~18.04.22_amd64.deb
Resolving security.ubuntu.com (security.ubuntu.com)... 91.189.91.38, 185.125.190.36, 91.189.91.39, ...
Connecting to security.ubuntu.com (security.ubuntu.com)|91.189.91.38|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2023-06-19 10:45:40 ERROR 404: Not Found.

------------------------------------------------------------------------------------------------
Error detected - status code 8
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer/ID1_ubuntu_os.sh, line 49
Stack Trace:
    [1] ID1_ubuntu_os(): ./installer/ID1_ubuntu_os.sh, line 49 -> ID1_ubuntu_os
    [2] main(): ./installer.sh, line 266 -> main -d

I also proceeded to check the link and it says the page is not found, this then halts the installation and proceeds to give me another error:

MongoDB - grep: /etc/mongod.conf: No such file or directory
not ok
    Wrong mongodb config - check your installation
    RE-run installation - bindIp should be set to 172.36.0.1
    Docker Interface - Device "emba_runs" does not exist.
    Missing Docker-Interface - check your installation
    Trying to auto-maintain the docker interface ...
Device "emba_runs" does not exist.
    Use  $systemctl restart NetworkManager docker or reset the docker interface manually ($ docker network rm emba_runs)
    cve-search - testing
[*] CVE-search not working - restarting Mongo database for CVE-search
Failed to restart mongod.service: Unit mongod.service not found.
[*] CVE-search not working - restarting Mongo database for CVE-search
Failed to restart mongod.service: Unit mongod.service not found.
    cve-search - not ok
[-] The needed CVE database is not responding as expected.
[-] CVE checks are currently not possible!
[-] Please check the following documentation on Github: https://github.com/e-m-b-a/emba/issues/187
[-] If this does not help, open a new issue here: https://github.com/e-m-b-a/emba/issues
    inotifywait - ok
    notify-send - ok

Some dependencies are missing - please check your installation

To install all needed dependencies, run 'sudo ./installer.sh'.
Learn more about the installation on the EMBA wiki: https://github.com/e-m-b-a/emba/wiki/installation

EMBA installation failed
Error detected - status code 8
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer/ID1_ubuntu_os.sh, line 49
Stack Trace:
    [1] ID1_ubuntu_os(): ./installer/ID1_ubuntu_os.sh, line 49 -> ID1_ubuntu_os
    [2] main(): ./installer.sh, line 266 -> main -d

So how do i solve this issue?

m-1-k-3, 2023-06-19:
are you using the latest EMBA git?

m-1-k-3, 2023-06-24:
For me this looks as you are running an out of date installer. See also here: https://github.com/e-m-b-a/emba/blob/3cad050f802b1a5849e086b44d80d91a7deadeb6/installer/ID1_ubuntu_os.sh#L49

L49 has no libssl installation anymore. We moved this within the installer a while ago.

Please re-open the issue with further details if needed.

closed 2023-06-24 by m-1-k-3