e-m-b-a/emba #697: Display the analysis time in the final aggregator

n0x08, 2023-07-13:
**Describe the solution you'd like**
I would like to see the overall time it took to analyze the firmware image displayed in the final aggregator page. Per-module analysis time could be good too as it would help me tune which modules to exclude from my scans when I want to process many images.

**Describe alternatives you've considered**
Currently I use 'time <emba command>'

**Priority issue**
Low priority

**Additional context**
An example of how this could look:
[+] Final aggregator
[+] Tested firmware: /home/nate/Polycom_3111-48820-021.sip.ld
[+] EMBA start command: ./emba -f /home/nate/Polycom_3111-48820-021.sip.ld -l /home/nate/Polycom_3111-48820-021 -p ./scan-profiles/default-scan.emba
[+] Detected architecture and endianness (verified): ARM / EL
 
[+] Operating system detected (verified): Linux / v4.19.0
[+] Total analysis time: 804 minutes

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-07-14:
Just to ensure I understand you correct. You need the testing time in the final report? I'm asking as the scan time is already shown in multiple areas:

[image]

You can access this information also from the web reporter:

[image]

[image]

n0x08, 2023-07-14:
Ooops :) I should have looked around more, that's exactly what I was looking for! Thank you!

m-1-k-3, 2023-07-24:
I think we can close this. If not, please re-open it again.

closed 2023-07-24 by m-1-k-3