e-m-b-a/emba #716: Install errors

torabi12, 2023-07-23:
**Describe the bug**
During the installation I got some warnings and errors. See screenshot section.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (default mode)
2. See error

**Expected behavior**
404 should be 200.

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.02
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
I wanted to update the previous emba installation and ./emba_updater was not working fine on my side, after this I made a new clean installation and I got the above messages.

m-1-k-3, 2023-07-23:
EMBA is trying multiple libssl download possibilities. Are all these tries resulting in a 404 or is it working on some stage? It shoudl try version 18.04.21 till 18.04.29. Is the installation finished and working?

torabi12, 2023-07-23:
EMBA updater drops this:

[image]

The installation went at the end and the scanning looks fine, but I just started a few minutes before.

m-1-k-3, 2023-07-23:
The updater looks not good

torabi12, 2023-07-23:
I found the updater error first, and after that I did a clean install and just wanted to mention the warnings, but the scan is running fine now.

m-1-k-3, 2023-07-24:
looks good on a fresh install

[image]

closed 2023-07-24 by m-1-k-3

torabi12, 2023-07-24:
The update script still drops the same error for me:

[image]