e-m-b-a/emba #725: The NVD plans to retire the remaining legacy data feeds as well as all 1.0 APIs.

m-1-k-3, 2023-07-29:
[CVE-Search](https://github.com/cve-search/cve-search) is our main tool for the CVE lookup functionality in the F20 module. These results are then the base for further lookups to Metasploit, Exploit-DB and other PoC databases. 

As NIST will retire the legacy data feeds in September we need to evaluate the consequences for EMBA:

https://nvd.nist.gov/General/News/change-timeline

[image]

Currently I'm not sure how hard this will hit cve-search and as a consequence also EMBA. It looks as cve-search are working on this [here](https://github.com/cve-search/vulnerability-lookup).
Currently I do not know if we have the possibility to further use and update cve-search after September.

Someone out there with more insights on this topic? Probably someone already tested [vulnerability-lookup](https://github.com/cve-search/vulnerability-lookup) and can share his experience and some results?

m-1-k-3, 2023-09-02:
[image]

a bit more time ...

m-1-k-3, 2023-09-14:
https://github.com/cve-search/cve-search/issues/929
Evaluate https://github.com/binareio/FastCVE
Evaluate https://github.com/cve-search/vulnerability-lookup/issues/1
https://github.com/fkie-cad/FACT_core/pull/1083

m-1-k-3, 2023-10-11:
Looks as CVE-search is getting a new importer https://github.com/cve-search/cve-search/pull/1010#issuecomment-1755654687

P-T-I, 2023-10-11:
@m-1-k-3 The importer is provided by the [cve-search/CveXplore](https://github.com/cve-search/CveXplore) package; feel free to check it out and post me the comments!

m-1-k-3, 2023-11-17:
Working on this. See https://github.com/e-m-b-a/emba/pull/884 and https://github.com/e-m-b-a/emba/pull/887
Further testing needed

m-1-k-3, 2023-11-18:
see also https://github.com/e-m-b-a/emba/pull/892

m-1-k-3, 2023-11-21:
should be done by https://github.com/e-m-b-a/emba/pull/899

m-1-k-3, 2023-11-21:
See now https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites
It is now recommend to setup a NIST API key

m-1-k-3, 2023-11-24:
Reopen it until final solution is finished. Currently working on a full replacement

closed 2023-11-28 by m-1-k-3