e-m-b-a/emba #734: why sbom file has not cpe and purl item？

chenjianquan7, 2023-08-03:
**Describe the bug**
i use emba analysis firmware , result an sbom json file ,but it has not cpe and purl item, why?
only name and version.
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: <Insert URL>
3. Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>
4. additional steps
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.01
- EMBA version: v1.2.2 or current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-08-03:
Currently no one has implemented it ;) PR are welcome

m-1-k-3, 2024-04-06:
problem solved? Anything else we can support you?

closed 2024-04-08 by m-1-k-3