e-m-b-a/emba #735: it return error "ImportError: cannot import name 'ContextAPI' from 'docker'" when install

chenjianquan7, 2023-08-03:
**Describe the bug**
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: <Insert URL>
3. Start EMBA with the following parameters: sudo ./emba.sh <insert further parameters>
4. additional steps
5. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.01
- EMBA version: v1.2.2 or current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [Y/N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-08-03:
Can you provide more details?

m-1-k-3, 2023-08-13:
any updates? Otherwise it will get closed.

closed 2023-08-21 by m-1-k-3