e-m-b-a/emba #742: emba update drops errors

torabi12, 2023-08-07:
**Describe the bug**
After a clean installation of the latest emba I tried the emba_updater script but it did not work for me. Please see the screenshot of it.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. Reboot
3. Start emba_updater script as root
4. See error

**Expected behavior**
DB update should start without any error.

**Screenshots**
[image]

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.02
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

BenediktMKuehne, 2023-08-09:
Had the same thing happening yesterday, installation succeeded though. And the dependency check does too.

torabi12, 2023-08-09:
I tried it for the 3rd time and I got the same error at the end of installation, CAPEC download was not working and after the reboot the update script was not OK.

m-1-k-3, 2023-08-10:
Looks like some issue with the new pipenv environment

m-1-k-3, 2023-08-10:
The CAPEC error should be fixed on cve-search side here https://github.com/cve-search/cve-search/releases/tag/v4.2.2 -> we need to update it soon

m-1-k-3, 2023-08-13:
The newest cve search version should be used now. Please check it and report if there are any issues. I will dig deeper into the update issue as soon as I'm back.

torabi12, 2023-08-13:
Hello,

I modified the line 40 in IF20_cve_search.sh to get the actual cve-search:
`git clone https://github.com/cve-search/cve-search.git external/cve-search`

and did a default installation. The installation went well:

[image]

and the CAPEC also downloaded:
[image]

I started a scan which will take some time but the beginning looked fine:
[image]

Please let me know if there is a better way to test the cve-search with e-m-b-a.

m-1-k-3, 2023-08-13:
The installer should already use the latest cve search. Give it a try without your modifications

torabi12, 2023-08-13:
The installation works fine in the untouched state.

PR #749 (2023-08-20): Fix updater
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

updater not fully working - see issue #742 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #742
closed 2023-08-20 by BenediktMKuehne