e-m-b-a/emba #743: CVE search is not working while using EMBA, although working while running separately.

shivamprakhar, 2023-08-08:
**Describe the bug**
I recently installed this tool and I am running in this issue while doing an analysis. The cve search options doesn't work while running emba but if I run it directly it works without any issues.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation (dev-mode/default mode)
2. Use the firmware available here: <Insert URL>
3. Start EMBA with the following parameters: sudo ./emba -l ~/Documents/ -f ~/Documents/ -p ./scan-profiles/full-scan.emba
4. additional steps
5. The needed CVE database is not responding as expected.
6. Whereas if I run the following command: ./external/cve-search/bin/search.py -p busybox , it runs without any issue.
7. I also tried some fix, sudo ./sbin/db_updater.py -c it throws another error

**Expected behavior**
For cve search to run properly.

**Screenshots**
[image]
[image]
[image]

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux latest
- EMBA version: current master branch
- Installation method: classic installation, no docker

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

BenediktMKuehne, 2023-08-10:
seems related to #742

shivamprakhar, 2023-08-10:
> seems related to #742

nope it's different than that issue, that issue looks like it's based on mission python package named 'nested_lookup'.

m-1-k-3, 2023-08-11:
Is it a fresh install with the current git Master?
Could you please check your external directory for the presence of the directory emba_venv?

shivamprakhar, 2023-08-12:
Yeah it was a fresh install, also I removed the external directory and tried to reinstall who everything again. But cve-search doesn't work.

m-1-k-3, 2023-08-12:
Have you loaded the EMBA python env? Should be something like 
source ./external/emba_venv/bin/activate 

You need todo this before you are manually working with the cve stuff

shivamprakhar, 2023-08-19:
Sorry for the late response, manual cve-search was fine, it didn't work when I was running EMBA scan. At the end I just removed everything and installed on a different system and it worked properly.

m-1-k-3, 2023-08-19:
Thank you for the response. I will close this issue now.

closed 2023-08-19 by m-1-k-3