e-m-b-a/emba #767: Kali Linux 2023.03 is out now

m-1-k-3, 2023-09-01:
New Kali Linux is out: https://www.kali.org/blog/kali-linux-2023-3-release/

We need to test EMBA on it

**Testcases:**

Testfirmware: [DLink DIR300](https://ftp.dlink.de/dir/dir-300/archive/driver_software/DIR-300_fw_revb_214b01_ALL_de_20130206.zip)

- [ ] Default/docker installation working with current docker image (`./installer.sh -d`)
  - [ ] dependency check (`./emba.sh -d 1`)
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)
- [ ] Full installation working (`./installer.sh -F`)
  - [ ] dependency check (`./emba.sh -d 1`)
  - [ ] EMBA run in dev mode with options -s -z -S -D -E -t -W -Q
  - [ ] EMBA run in dev mode with options -s -z -S -D, -c, -E, -t, -W -Q
- [ ] Docker build (`sudo docker-compose build --no-cache --pull`)
  - [ ] dependency check (`./emba.sh -d 2`)
  - [ ] EMBA run with profile default-scan in strict mode (-S)
  - [ ] EMBA run with profile default-scan-emulation in strict mode (-S)
  - [ ] EMBA run with profile full-scan in strict mode (-S)

torabi12, 2023-09-01:
I would do the default/docker installation and tests on the weekend with the latest Kali.

torabi12, 2023-09-03:
- Installation was OK with the current master release.
- Dependency check gave me this message more times:

[image]

- default-scan in strict mode (-S) started but gave back back prompt at the beggining with this message:

[image]

This was the init command:
`$ sudo ./emba -l ~/DIR300_default -f ~/DIR300B5_FW214WWB01.bin -p ./scan-profiles/default-scan.emba -S -X 2.14 -Y D-Link -Z DIR-300 -N DIR300B5_FW214WWB01.bin`

Could you help what to try for better results?

m-1-k-3, 2023-09-03:
this is the messaging system. Looks like something changed under the hood. I need to take a look at this. Is the scan starting up in "non-strict" mode (remove -S option)?

torabi12, 2023-09-03:
Without -S the scan has started. It displayed the same errors but went on and now it is running....

m-1-k-3, 2023-09-03:
ok, this means a normal EMBA user is not directly affected. I will check on this

torabi12, 2023-09-03:
Default scan is finished, report looks fine. The above error came at the end also:
[image]

I would wait for any updates for dbus-launch message with the next scans.

torabi12, 2023-09-03:
> Looks like we need to install the following package:
> 
> ```
> sudo apt install dbus-x11
> ```
> 
> Can you verify this on your installation?

With dbus-x11 package the dependency check was OK without any error, and the strict-mode default scan has started also now.
I am going to execute the remaining tests on this environment.

torabi12, 2023-09-03:
I executed the default tests, all in strict mode.

- default-scan
[image]

- emulation
[image]

- full
[image]

When the dbus-x11 will be installed with EMBA I will install a new clean release again, but I didn't see other issue under the setup.

m-1-k-3, 2023-09-06:
Another fix for the dev mode https://github.com/e-m-b-a/emba/pull/777

m-1-k-3, 2023-09-07:
thx for all your support. Last tests are finished!

closed 2023-09-07 by m-1-k-3