e-m-b-a/emba #775: Black Screen Issue When Scanning 300M Cabin Package in Docker

gaolXue, 2023-09-04:
**Describe the bug**
I'm attempting to scan a cabin package of 300M . the program appears to get stuck at the "yara check finished" step, causing my computer screen to go black so that I can only access it via SSH. Is it possible that the scanner cannot execute the hardware emulation when dealing with larger packages?how can I disable the hardware emulation feature? If hardware emulation is turned off, will it affect the ability to generate CVEs?
Thank you!
[image]
[image]

m-1-k-3, 2023-09-04:
Can you provide a download link from the firmware and your startup command. Additionally, you could check the system log with dmesg ... probably you are running out of memory.

gaolXue, 2023-09-04:
I'm sorry, but we cannot provide this firmware package. However, I can provide the following information:

We are using the default command for scanning.

Server Information:
CPU: 12th Gen Intel(R) Core(TM) i5-12500
8 cores, 32GB RAM, 2TB storage

Firmware Package Information:
Size: 300MB, contains multiple compressed files
After reversing with emba, it's approximately 7GB in size.

the system configuration is designed to minimize the chances of running into memory issues, I think. And  I will update the system log with dmesg later tonight.

Thanks for your support.

m-1-k-3, 2023-09-04:
Please check after such a blank screen with `dmesg` for OOM messages.

gaolXue, 2023-09-05:
updated on my side as below,
there is no error message in system log.
[image]
there is the tmp error information.
[image]
Is there any other logs that can help us locate the issue?   or can we just skip the scan model of l10_system_emulation which stuck the process.   if so, please provide the command to skip the model. 
if you need more information, please let me know.
thanks for you support.

m-1-k-3, 2023-09-05:
According your initial posted logs the module l10 is not started. As you have not provided your startup command I do not know if it should started or not. The logs are somewhere in the scanning phase.

You can check for not finished modules with the following helper script:
```
./helpers/running_modules.sh <Path to your EMBA log directory>
```

Additionally, It looks as you do not use any profile but without further details I do not know.

Are there any out of memory messages printed via dmesg?

Additionally, I would recommend a run in strict mode with parameter -S. Probably we can find an error log in the EMBA log directory afterwards.

m-1-k-3, 2023-09-14:
Could we have an update on this issue?

m-1-k-3, 2023-09-17:
reopen if needed

closed 2023-09-17 by m-1-k-3