e-m-b-a/emba #778: Error installing on Kali 2023

AnotherWayIn, 2023-09-07:
Tried to install multiple times - from both vm or full install of Kali Linux 2023.03
Every time it fails at the same spot:

`023-09-07 13:15:52,490 - CPEDownloads - INFO     - Finished CPE database population
2023-09-07 13:15:53,286 - CVEDownloads - INFO     - CVE database population started
2023-09-07 13:15:53,286 - CVEDownloads - INFO     - Starting CVE database population starting from year: 2002
2023-09-07 13:15:53,374 - DatabaseIndexer - INFO     - Success to create index [('id', 1)] on cves
2023-09-07 13:15:53,584 - DatabaseIndexer - INFO     - Success to create index [('vulnerable_configuration', 1)] on cves
2023-09-07 13:15:54,794 - DatabaseIndexer - INFO     - Success to create index [('vulnerable_product', 1)] on cves
2023-09-07 13:15:54,892 - DatabaseIndexer - INFO     - Success to create index [('Modified', 1)] on cves
2023-09-07 13:15:54,913 - DatabaseIndexer - INFO     - Success to create index [('Published', 1)] on cves
2023-09-07 13:15:54,936 - DatabaseIndexer - INFO     - Success to create index [('last-modified', 1)] on cves
2023-09-07 13:15:54,961 - DatabaseIndexer - INFO     - Success to create index [('cvss', 1)] on cves
2023-09-07 13:15:54,978 - DatabaseIndexer - INFO     - Success to create index [('cvss3', 1)] on cves
2023-09-07 13:15:55,003 - DatabaseIndexer - INFO     - Success to create index [('summary', 'text')] on cves
2023-09-07 13:15:55,024 - DatabaseIndexer - INFO     - Success to create index [('vendors', 1)] on cves
2023-09-07 13:15:55,044 - DatabaseIndexer - INFO     - Success to create index [('products', 1)] on cves
2023-09-07 13:15:55,064 - DatabaseIndexer - INFO     - Success to create index [('vulnerable_product_stems', 1)] on cves
2023-09-07 13:15:55,087 - DatabaseIndexer - INFO     - Success to create index [('vulnerable_configuration_stems', 1)] on cves
Downloading files: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 22/22 [00:29<00:00,  1.36s/it]
Processing downloaded files: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 22/22 [56:32<00:00, 154.22s/it]
Transferring queue to database:   8%|█████████▋                                                                                                              | 18078/223618 [00:18<04:32, 755.55it/s]Process Process-3:
Process Process-11:
Process Process-13:
Process Process-4:
Process Process-6:
Process Process-2:
Process Process-10:
Process Process-5:
Process Process-8:
Transferring queue to database:  10%|████████████                                                                                                            | 22423/223618 [00:23<04:20, 772.32it/s]Traceback (most recent call last):
Traceback (most recent call last):
Traceback (most recent call last):
Traceback (most recent call last):
Traceback (most recent call last):
Traceback (most recent call last):
Traceback (most recent call last):
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
Traceback (most recent call last):
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/usr/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/DownloadHandler.py", line 245, in _db_bulk_writer
    for batch in iter(lambda: list(islice(self.queue, threshold)), []):
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/DownloadHandler.py", line 245, in <lambda>
    for batch in iter(lambda: list(islice(self.queue, threshold)), []):
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/redis_q.py", line 23, in __next__
    item = self.get(timeout=1)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/redis_q.py", line 73, in get
    item = self.__db.lpop(self.key)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/sean/tools/emba/external/emba_venv/lib/python3.11/site-packages/redis/commands/core.py", line 2513, in lpop
    return self.execute_command("LPOP", name)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/sean/tools/emba/external/emba_venv/lib/python3.11/site-packages/redis/client.py", line 1218, in execute_command
    return conn.retry.call_with_retry(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/DownloadHandler.py", line 245, in _db_bulk_writer
    for batch in iter(lambda: list(islice(self.queue, threshold)), []):
  File "/home/sean/tools/emba/external/emba_venv/lib/python3.11/site-packages/redis/retry.py", line 45, in call_with_retry
[... 577 lines cut ...]
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sean/tools/emba/external/emba_venv/lib/python3.11/site-packages/redis/connection.py", line 336, in read_response
    raise error
redis.exceptions.BusyLoadingError: Redis is loading the dataset in memory

The cron.daily update script for EMBA is located in config/emba_updater
For automatic updates it should be checked and copied to /etc/cron.daily/
Cloning into 'external/trickest-cve'...
remote: Enumerating objects: 188879, done.
remote: Total 188879 (delta 0), reused 0 (delta 0), pack-reused 188879
Receiving objects: 100% (188879/188879), 166.16 MiB | 2.75 MiB/s, done.
Resolving deltas: 100% (152106/152106), done.
Updating files: 100% (75029/75029), done.

Installation notes:

INFO: The cron.daily update script for EMBA is located in config/emba_updater
INFO: For automatic updates it should be copied to /etc/cron.daily/
INFO: For manual updates just start it via sudo ./config/emba_updater

WARNING: If you plan using the emulator (-E switch) your host and your internal network needs to be protected.

INFO: Do not forget to checkout current development of EMBA at https://github.com/e-m-b-a.
EMBA installation finished 
`

m-1-k-3, 2023-09-07:
Is your system fulfilling the hardware [requirements](https://github.com/e-m-b-a/emba/wiki/Installation#prerequisites)?

AnotherWayIn, 2023-09-07:
x86 and 8gb ram

AnotherWayIn, 2023-09-07:
maybe a python 3.11.4 issue?

`sudo ./sbin/db_updater.py -c
Traceback (most recent call last):
  File "/home/sean/tools/emba/external/cve-search/./sbin/db_mgmt_cpe_dictionary.py", line 27, in <module>
    from lib.Sources_process import CPEDownloads
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/Sources_process.py", line 26, in <module>
    from lib.JSONFileHandler import JSONFileHandler
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/JSONFileHandler.py", line 4, in <module>
    from lib.DownloadHandler import DownloadHandler
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/DownloadHandler.py", line 26, in <module>
    from lib.LogHandler import UpdateHandler
  File "/home/sean/tools/emba/external/cve-search/sbin/../lib/LogHandler.py", line 11, in <module>
    import colors
ModuleNotFoundError: No module named 'colors'`

m-1-k-3, 2023-09-07:
I think you need to activate the python environment first:

```
source ./external/emba_venv/bin/activate
```

AnotherWayIn, 2023-09-07:
thanks. that didnt work :(

 Also, probably a seperate issue but there's something not right still with the installer as ive done another clean install and i get the following:

`└─$ sudo ./installer.sh -d  

EMBA - Embedded Linux Analyzer Installer
=================================================================

==> Imported 26 installer module files

Install all dependecies for EMBA in default/docker mode

Created external directory: ./external

Update package lists.
Ign:1 https://ftp.ne.jp/Linux/packages/kali/kali kali-rolling InRelease
Err:2 https://ftp.ne.jp/Linux/packages/kali/kali kali-rolling Release
  SRV records for ftp.ne.jp indicate that https service is not available at this domain
Reading package lists... Done
E: The repository 'https://http.kali.org/kali kali-rolling Release' no longer has a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
------------------------------------------------------------------------------------------------
Error detected - status code 100
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer.sh, line 266
Stack Trace:
    [1] main(): ./installer.sh, line 266 -> main -d
`

m-1-k-3, 2023-09-07:
ok, there is not https repo available. We probably need to set this via an option. More repos then expected are only available via http.

m-1-k-3, 2023-09-07:
The HTTPS issue should be fixed in PR #779. Regarding the redis issue (cve-search) I would first try to put more RAM into the box ;)

AnotherWayIn, 2023-09-13:
Thanks, i installed 16gb and had to fix some python dependencies for cve-search, now it works!

m-1-k-3, 2023-09-13:
Usually the python dependencies should be resolved from the installer. I'm wondering what is going wrong here. If can figure it out, please let me know

AnotherWayIn, 2023-09-13:
sorry, i'm not sure. the installer completed fine once i upped the ram, but ember failed cve search when it first ran. So i took the following steps. hopefully this will help: 

- python external/cve-search/bin/search.py - error no such module: nested_lookup, colors and other dependencies.
- source ./external/emba_venv/bin/activate 
- pip install -r external/cve-search/requirements.txt - failed on one dependency (visitor) due to permissions error
- sudo pip install visitor

then it worked!

closed 2023-09-17 by m-1-k-3