e-m-b-a/emba #780: Improve BIOS/UEFI support

m-1-k-3, 2023-09-09:
**Is your feature request related to a problem? Please describe.**

EMBA is used in a lot of different areas. One of these areas is around BIOS and UEFI analysis. See here: https://www.basicinputoutput.com/2023/08/software-bill-of-materials-sbom-in.html

Probably we can improve the results in the future. 

For this we need your support! Please post download links to UEFI and BIOS firmware that we can use for further analysis.

torabi12, 2023-09-09:
Server platform might be interesting, this Dell BIOS contains a cap file for testing:
https://dl.dell.com/FOLDER10220002M/1/BIOS_W3X7P_WN64_2.19.1.EXE

Asus is inevitable BIOS provider I think:
https://dlcdnets.asus.com/pub/ASUS/mb/BIOS/PRIME-B560M-K-ASUS-2001.zip

HP is also a huge manufacturer with regular updates on BIOS/FW area:
https://ftp.hp.com/pub/softpaq/sp148001-148500/sp148108.exe

My only experience is that testing is very long if I'm testing UEFI BIOS with EMBA, you can expect to run practically 1 day or more with default settings.

m-1-k-3, 2023-11-08:
@torabi12 can you provide an EMBA report and the related UEFI firmware from such a long running UEFI test? Probably offline?

torabi12, 2023-11-08:
I did a scan for the initial version of a Dell server BIOS:
https://www.dell.com/support/home/hu-hu/drivers/driversdetails?driverid=dt5jm&oscode=biosa&productcode=poweredge-r440

[html-report.zip](https://github.com/e-m-b-a/emba/files/13298547/html-report.zip)

but it was just a short test:
Test ended on Wed Nov  8 16:29:49 CET 2023 and took about 0 days and 00:40:04 

I used proxy under this, do you need an offline scan?

m-1-k-3, 2023-11-10:
@torabi12 if you find some time for testing the improved UEFI handling, your feedback would be very interesting for us :)

torabi12, 2023-11-10:
> @torabi12 if you find some time for testing the improved UEFI handling, your feedback would be very interesting for us :)

Thanks, I would start a new scan, but I am using the actual master release and I don't know how to move to m-1-k-3:uefi_improvements without reinstalling the system. If you could help it would be faster. :)

m-1-k-3, 2023-11-10:
probably the easiest way is to clone the branch into a separate directory and link the external directory to your current installation:

```
git clone --branch uefi_improvements https://github.com/m-1-k-3/emba.git emba_test
cd emba_test
ln -s /PATH_TO_YOUR_EXISTING_EMBA_INSTALLATION/external ./external
```

This should be enough :)

torabi12, 2023-11-10:
Thank you, I started a scan for an older Asus motherboard:
https://dlcdnets.asus.com/pub/ASUS/mb/LGA1151/H110M-A/H110M-A-ASUS-3805.zip?model=H110M-A

Last time this scan was quite long. Tomorrow we will see:

[image]

My OpenAI API key is not the best, it used to work but nowadays is drops quota limit issue:(

m-1-k-3, 2023-11-10:
as long as you are going to test the cap and not the zip it should work quite well ;)

torabi12, 2023-11-10:
I used the CAP file for emba, only the download link contains the ZIP format.

torabi12, 2023-11-11:
Asus BIOS scan was not so long:
[image]

and the report:
[H110M-A-ASUS-3805_html-report.zip](https://github.com/e-m-b-a/emba/files/13324852/H110M-A-ASUS-3805_html-report.zip)

m-1-k-3, 2023-11-11:
looks quite good to me. :) Thank you for testing
We need to include it into the deep extractor. With this in place we can also handle archived firmware in the future

torabi12, 2023-11-11:
Thanks for the help with the test.

closed 2023-11-13 by HoxhaEndri