e-m-b-a/emba #786: Q02_openai_question

torabi12, 2023-09-12:
**Describe the bug**
I did a scan with using default-scan-gpt.emba profile and I didn't get any OpenAI answer, the q02_openai_question.txt file includes only this:
Q02_openai_question nothing reported

I used a classic FW file with many vulnerabilities:
[image]

The GPT Resolver section started like this:
[image]

Before the scan I set option 1:
`$ export GPT_OPTION=1`

and the gpt_config.env file uncommented like this:
[image]

I ran through the html report but I didn't find OpenAI related page or info on the pages.
Could you help me how to do a better scan which provides ChatGPT answers in the final report?

This was the command for the scan:

`$ sudo ./emba -l ~/Kankun -f ~/kkeps.bin -p ./scan-profiles/default-scan-gpt.emba -X 1.0 -Y Kankun -Z SmartPlug -N kkeps.bin`

**To Reproduce**
1. EMBA installation (default mode)
2. Use the firmware available here: https://github.com/jdpiercy/Kankun-SmartPlug/blob/master/Firmware%20-%20Stock/kkeps-3.1.bin
3. Start EMBA with the following parameters: `$ sudo ./emba -l ~/Kankun -f ~/kkeps.bin -p ./scan-profiles/default-scan-gpt.emba -X 1.0 -Y Kankun -Z SmartPlug -N kkeps.bin`

**Screenshots**
See above.

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.03
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

m-1-k-3, 2023-09-13:
Some notes on debugging:
* EMBA is currently using GPT only for selected modules like the shell scripting, python, php and lua module. Are there any scripts like this in your firmware?
* If you use the gpt scan profile you do not need to export the GPT variable on cli.
* Is EMBA able to reach the Internet? See dependency check in the beginning
[image]
* You can monitor the quest container during runtime with docker ps and docker logs

torabi12, 2023-09-13:
Thank you, I did another scan for a different FW and I got the answers:

[image]

closed 2023-09-13 by torabi12