e-m-b-a/emba #805: Debian11.7.0 issues

V33RU, 2023-09-29:
Errros encountered while installing emba  in debian 11 ,

EMBA docker image will be downloaded.
CONTAINER VARIABLE SET TO embeddedanalyzer/emba:latest
latest: Pulling from embeddedanalyzer/emba
Digest: sha256:a4bf71cad91063208c5e8b163fd885960c9c9ba29e8507b68ddb3a62cc22a2b6
Status: Image is up to date for embeddedanalyzer/emba:latest
docker.io/embeddedanalyzer/emba:latest
WARNING: The FIRMWARE variable is not set. Defaulting to a blank string.
WARNING: The LOG variable is not set. Defaulting to a blank string.
WARNING: The EMBA variable is not set. Defaulting to a blank string.
ERROR: The Compose file './docker-compose.yml' is invalid because:
networks.emba_runs value Additional properties are not allowed ('name' was unexpected)
------------------------------------------------------------------------------------------------
Error detected - status code 1
Command:  echo -e "Command:  $ORANGE$BASH_COMMAND$NC"
Location:  ./installer/I05_emba_docker_image_dl.sh, line 55
Stack Trace:
    [1] I05_emba_docker_image_dl(): ./installer/I05_emba_docker_image_dl.sh, line 55 -> I05_emba_docker_image_dl
    [2] main(): ./installer.sh, line 310 -> main -d

Important: Consider filling out a bug report at https://github.com/e-m-b-a/emba/issues

m-1-k-3, 2023-09-29:
Thanks for reporting. Could this be the same issue reported here: https://github.com/e-m-b-a/emba/issues/247

alexandruchirita4192, 2023-10-01:
Hi,

I wanted to add that I was able to run `emba` on Debian 12.1 with the following commands:
```
git clone https://github.com/e-m-b-a/emba.git
cd emba
sudo ./installer.sh -d
sudo ./emba -l ~/log -f /lib/firmware -p ./scan-profiles/default-scan.emba
```

Maybe you could upgrade your Debian version and retry running emba?

Screenshot to prove that it run successfully on Debian 12.1:
[image]
[image]

Screenshot to prove that results are given, I also found that `nvidia` firmware is insecure:
[image]

Good luck!

V33RU, 2023-10-03:
Ok using latest version of debian, thank you ..

closed 2023-10-03 by V33RU