e-m-b-a/emba #823: Could not install libssl

torabi12, 2023-10-10:
**Describe the bug**
I started a new installation but it stopped here:

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. See error

**Expected behavior**
Go on to get the final app.

**Desktop (please complete the following information):**
- OS: Kali Linux 2023.3
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
Add any other context about the problem here.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-10-10:
could you give it a try https://github.com/m-1-k-3/emba/tree/echo_if20

torabi12, 2023-10-10:
I used this URL:
https://github.com/m-1-k-3/emba.git

but I got the same stop. Or may I try something else? :(

torabi12, 2023-10-10:
> I used this URL: https://github.com/m-1-k-3/emba.git
> 
> but I got the same stop. Or may I try something else? :(

I just wanted to say that I was not sure I tried your request. If I made something wrong please let me know. Thank you.

m-1-k-3, 2023-10-11:
Please use the branch _[echo_if20](https://github.com/m-1-k-3/emba/tree/echo_if20)_:

```
git clone --branch echo_if20 https://github.com/m-1-k-3/emba.git
```

torabi12, 2023-10-11:
Thank you. Unfortunatelly I got the same:
[image]

m-1-k-3, 2023-10-11:
is it not checking nr 23?

[image]

torabi12, 2023-10-11:
Both files are downloaded but the installer stops after 18.04.29_amd64.dep package:

[image]

m-1-k-3, 2023-10-11:
Currently I am not able to reproduce this issue:

[image]

torabi12, 2023-10-11:
Pff, I got a bigger SSD a few days before and I wanted to install a brand new Kali 2023.3 with latest updates and an actual emba. On my side it stops for sure:(

[image]

[install.log](https://github.com/e-m-b-a/emba/files/12871790/install.log)

m-1-k-3, 2023-10-11:
your new SSD boes brrrrm ;)

please check my dev branch again.

torabi12, 2023-10-11:
Sorry but it stopped again at the same point.

I took the Kali ISO for VMware, I just extended the default 80GB to a bigger one with gparted.
I created an own user, which belongs to the sudo group as well. Chenged the default hostname and ZSH to bash and my timezone:

[image]

and I updated the system:

```
$ sudo apt update
$ sudo apt purge openvas-scanner
$ sudo apt purge crackmapexec
$ sudo apt upgrade -y
$ sudo apt install ssh numlockx mc htop -y
```

and started the emba installation.

Sorry for this, maybe I made a mistake somewhere, but I don't know what did I wrongly:)
SSH is enabled, but I think no other major modification on the VM.

m-1-k-3, 2023-10-12:
can you try to download the libssl deb files manually with wget and check them with the file command?
So, please do something like 

```
wget http://security.ubuntu.com/ubuntu/pool/main/o/openssl/libssl1.1_1.1.1-1ubuntu2.1~18.04.23_amd64.deb -O external/libssl.deb
file external/libssl.deb
```

Thank you

torabi12, 2023-10-12:
The manual download works fine:

[image]

The installer downloads with HTTP 200, but the file is not there. I'm going crazy. :)

m-1-k-3, 2023-10-12:
pls try the dev branch again ...

torabi12, 2023-10-12:
I went back to clean snapshot and used this:
`$ git clone --branch echo_if20 https://github.com/m-1-k-3/emba.git`

This is on the screen right now:

[image]

m-1-k-3, 2023-10-12:
Sorry but I don't see enough details on this screenshot. I need the whole libssl process till the stop

torabi12, 2023-10-12:
> Sorry but I don't see enough details on this screenshot. I need the whole libssl process till the stop

[libssl.txt](https://github.com/e-m-b-a/emba/files/12878538/libssl.txt)

Please let me know if you need something else. Thank you very much.

m-1-k-3, 2023-10-12:
can you gimme your version of file:
```
└─$ file --version
```

torabi12, 2023-10-12:
```
$ file --version                                                                                                                                                                                                      
file-5.45
magic file from /etc/magic:/usr/share/misc/magic
```

m-1-k-3, 2023-10-12:
check dev branch again pls

torabi12, 2023-10-12:
wow, now it went on:

[image]

What was wrong? What did I wrongly?
Thank you very much!

m-1-k-3, 2023-10-12:
great ... the new file version has changed the output of the debian package:

file v5.45:
```
Debian binary package (format 2.0), with control.tar.xz , data compression xz
```
file < v5.45:
```
Debian binary package (format 2.0), with control.tar.xz, data compression xz
```

got it?

Thanks for reporting and your effort in working with us on a fix
Good job :)

PR #826 (2023-10-12): Installer debian package file format
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

see issue #823 

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

closes #823
closed 2023-10-12 by m-1-k-3