e-m-b-a/emba #825: Rule scanning

gaolXue, 2023-10-12:
**Describe the bug**

1. Now we are scanning a firmware package in the car, emba found about 50,000 files in it, EMBA has been stuck in P60 module for nearly 20 days, we want to know whether the P60 module can be removed
2. Whether EMBA supports MISRA static scanning rules

Thank you!

m-1-k-3, 2023-10-12:
1.) P60 is the deep extractor and it tries to extract everything multiple times. It is more or less the heart of the EMBA extraction mechanisms. This means it can't be removed.
If you can extract the firmware manually, then extract it to a directory and point EMBA to this directory with the -f parameter.

2.) Currently MISRA code scanning is not implemented as EMBA usually has no source code available. If you have a firmware and source code feel free to open a PR with a MISRA module.

m-1-k-3, 2023-12-09:
Probably we can use the Ghidra results from PR #929  also for this MISRA code scanning. I will close this issue and we can start working on #929 first.

closed 2023-12-09 by m-1-k-3