e-m-b-a/emba #839: Files not found in report

torabi12, 2023-10-20:
**Describe the bug**
At the bottom of the report's menu under + Show more results the stacs_pw_hashes and common_linux_files pages are not available. Bot drops the same file not found errors in the browser.

Firefox can’t find the file at /home/seclab/Kankun/html-report//etc/stacs_pw_hashes.html.
Firefox can’t find the file at /home/seclab/Kankun/html-report//etc/common_linux_files.html.

**To Reproduce**
Steps to reproduce the behavior:
1. EMBA installation default mode
2. Use the firmware available here: kkeps.bin but any other should give the same result
3. Start EMBA with the following parameters: $ sudo ./emba -l ~/Kankun -f ~/kkeps.bin -t -s -z -W -X 1.0 -Y Kankun -Z Smartplug -N kkeps.bin
4. Wait the end of the scan
5. Check the "Show more results" section in the report

**Expected behavior**
Pages should be displayed, or if not applicable report' menu should not contain these items.

**Screenshots**
[image]

The same scan in EMBArk is also not OK:

[image]

**Desktop (please complete the following information):**
- OS: e.g. Kali Linux 2023.3
- EMBA version: current master branch
- Installation method: default with up to date docker image

**Priority issue**
Are you already a [Sponsor]? - [N]

**Additional context**
None.

[Sponsor]: https://github.com/sponsors/e-m-b-a

m-1-k-3, 2023-10-21:
Is this issue something new or is it already for a while in EMBA?

torabi12, 2023-10-21:
It used to work. I checked some older reports and for example on 09.14. was working. Both pages were available in the report, but I don't know when it went bad.
I noticed it by accident a few days ago, because I've run a lot of tests now and all of them have these two pages wrong.

m-1-k-3, 2023-10-22:
same here ... 
[image]

PR #844 (2023-10-23): fix #839
* **What kind of change does this PR introduce?** (Bug fix, feature, docs update, ...)

Bug fix

* **What is the current behavior?** (You can also link to an open issue here)

irrelevant files are linked but missing

* **What is the new behavior (if this is a feature change)? If possible add a screenshot.**

feature removed - closes #839
commit 1fb8acff (2023-10-23): fix #839
commit c2b832a6 (2023-10-23): Merge pull request #844 from m-1-k-3/master

fix #839
closed 2023-10-23 by m-1-k-3